ISS-Mimic/Mimic #14: Expected fight between LED's and interrupts -> Undetected joint angle error

juggler2000, 2017-01-28:
The LED libraries (both Adafruits and FastLED) temporarily disable interrupts when sending commands to the LED strips.  This is because the timing on the embedded control chips is very sensitive.

The potential issue is that we could miss some encoder-provoked interrupts while transmitting to the LEDs, meaning we could miss some encoder counts.  This would cause us to rotate the motors further than desired, and never know it.

It's not a big problem in the near-term, as the error would likely be small (although it would accumulate), particularly if we don't update the LED commands very often. 

Potential solutions:
1) Command all motors to stop, and wait for them to do so, prior to updating the LEDs.  This is only feasible if we would rarely update the LEDs.
2) Estimate how much we have traveled while the interrupts were disabled based on the speed command, and update our counts accordingly.  
3) Offload the actual commanding of LED's to a secondary arduino, dedicated to sending only LED commands.  This would require some kind of communication between the primary (Mega) and secondary.  Fortunately, the Mega has an unused hardware serial that could be used for this (or software serial or other protocol on other pins).  But this would add complexity to it all.
4) If we do get some kind of auto-calibration going, this could update joint angles as needed.  This means we'd be "wrong" most of the time, and only "right" immediately after calibration.  Our current calibration plan would only work when the SARJ is at one of two angles, and may require BGA's to be at particular angles as well.  So this calibration would be either infrequent or we'd have to think of an alternative approach.

juggler2000, 2017-01-28:
I guess an additional option is to command the LED's directly from the Pi.  But this would require parsing and extra work for the Pi, which would be nice to avoid.

streadgold, 2017-02-06:
Personally I like option 3, seems easiest and least risky to motor position error. Although there is a python wrapper for the neopixel library... so maybe commanding them directly from the Pi would be easier and we'd have to keep track of less data on the arduino side...

ISS-Mimic, 2017-02-13:
Commanding from Pi will not work - use of Neopixel library on Pi requires root access for devmem, root access (aside from being dangerous for an internet connected device) currently render GUI unusable for other reasons (freezes when fetching JSON info for crew screen).

Option #3 now seems likeliest/easiest to implement.

juggler2000, 2018-01-08:
Here's some Kung fu that may allow a single M0 ard to handle the motors and the pixels.  And, it does so by utilizing one of the SPI peripherals (magic).  And, it can do all of this on some bonus pins hiding in the middle of the board, ostensibly also providing a bag of chips.

https://learn.adafruit.com/dma-driven-neopixels/overview

ISS-Mimic, 2018-02-27:
Secondary arduino for LEDs seems to work just fine

closed 2018-02-27 by ISS-Mimic