ISS-Mimic/Mimic #32: Arduino loop time toggles between 7 and 615ms

juggler2000, 2017-10-28:
For some reason, the Arduino Loop time (start to finish within loop function) is either 7ms (yea!) or 615 ms (boo) with no known cause.  It's never anything in between.  It doesn't seem to matter if the motor shield is powered externally or not.

I couldn't find any floating ground or other impropriety.  There doesn't seem to be anything in the loop that would cause the system to wait for anything.  Serial comm with PC is at high rate (115k).

Hitting the reset button seems to eventually put it into the slower state.  Sometimes, unplugging and re-plugging the USB from the PC gets you back to the fast mode.

My only thought would be something with the interrupts, or perhaps demon possession.

Here's an example of it going from swell to hell.  All I did was press the reset button on the motor shield.  The serial output says "Motor test" a that point.  The last value on each line is the loop duration, in ms.  It's also obvious from the rate of serial prints to the screen that it's taking much longer.

190.00,32.40,250,6714.67|,190.00,33.40,250,3540.57, deltaT:7
190.00,33.20,250,6690.51|,190.00,34.00,250,3536.37, deltaT:7
190.00,34.00,250,6666.31|,190.00,34.80,250,3528.13, deltaT:7
190.00,34.80,250,6642.07|,190.00,35.60,250,3519.85, deltaT:7
190.00,35.40,250,6625.80|,190.00,36.40,250,3511.53, deltaT:7
190.00,36.20,250,6601.49|,190.00,37.20,250,3503.17, ⸮Motor test!
190.00,143.20,250,1872.00|,190.00,139.20,250,1016.00, deltaT:617
190.00,191.40,55,-56.00|,190.00,186.20,87,87.78, deltaT:616
190.00,239.40,250,-2128.89|,190.00,233.20,250,-864.00, deltaT:616
190.00,288.00,250,-4376.20|,190.00,279.80,250,-2073.93, deltaT:616
190.00,336.60,250,-6773.93|,190.00,326.40,250,-3428.09, deltaT:616
190.00,374.00,250,-8839.41|,190.00,360.40,250,-4635.48, deltaT:615
190.00,380.20,250,-9675.13|,190.00,365.00,250,-5268.23, deltaT:616

juggler2000, 2018-01-01:
I asked for help on Adafruit's forum, and Adafuit staff essentially said that the Due is problematic and suggested avoiding it.  They pointed me to an M0-based board to get the number of interrupts we need without the headaches of the Due, so I ordered one of their "M0 Metro" boards and should have it later this week.

https://forums.adafruit.com/viewtopic.php?f=31&t=128661

SupaTreadz, 2018-01-02:
That would also resolve issue #32a "No one knows how to pronounce Due"

juggler2000, 2018-01-08:
Using Arduino Metro M0 with aforementioned motor shield, no longer having any of the timing problems.  Working perfectly with 3 DC motors and 1 servo.  Will try with more motors soon, but this ~~solves~~ abandons the timing problem by moving to M0 board.

Issue resolved by murdering Due and moving to M0-based board.

closed 2018-01-08 by juggler2000