ISS-Mimic/Mimic #35: Passing data (joint angles, etc) from RasPi to Ard

juggler2000, 2018-01-08:
I was thinking one way to pass data for joint angles, etc, is via the .json standard, which uses named pairs: name, then value.  We could perhaps use an existing json library, like this: https://arduinojson.org/

SupaTreadz, 2018-01-08:
Can do! Python makes JSON wrapping super easy

juggler2000, 2018-01-15:
Leaning towards this Ard Json library, as it appears most widely used.  Currently have not clue how to use it.
https://arduinojson.org/

juggler2000, 2018-02-05:
Does the Pi send data over serial by default, or do I need to hit "transmit".  GUI seems to lock up every time I hit transmit (the GUI is unresponsive, requiring hard reboot).

streadgold, 2018-02-05:
If the Pi is locking up that means the Arduino isn't connected or it isn't detecting the Arduino. When booting the Pi, along with the kivy loading stuff it should list the connected serial port if there is one. Also in the error log file (in logs folder) it says if a successful connection occurred, if there are three serial connection errors listed it didn't connect to anything. 

This is connecting one of the Pi USB ports to the Arduino USB port right? Try another USB port on the Pi?

streadgold, 2018-02-05:
Also I've only managed to get it working with USB to USB, so debugging might be tricky. Still working on USB to TX/RX pins

juggler2000, 2018-02-05:
I forgot it would lock up w/o connection.  Iwas being cruel, and just had
serial pins going from Pi usb to O-scope.  Now trying the TX/RX pins (via
logic level shifting).  We'll see.

streadgold, 2018-02-07:
Just FYI - I just tested my Mega connected to the Pi over USB and outputting serial to the computer over the TX1/RX1 pins. The code is in the Arduino folder. Not sure what needs to change for that to work on the your board but the concept works fine.

ISS-Mimic, 2018-02-27:
Closing, I believe our tests confirm we have this problem down

closed 2018-02-27 by ISS-Mimic