ISS-Mimic/Mimic #36: Convert buttons to graphics

streadgold, 2018-01-24:
Most GUI buttons are still the basic kivy widgets, need to replace them with custom graphics that have depth and can be depressed (replaced with depressed button graphic on_press)

GodspeedDiscvry, 2018-03-08:
I believe I have provided all of these graphics.

ISS-Mimic, 2018-03-08:
For default state yes, but this is for both unpressed and pressed buttons, so every button will need two graphics (for fanciness). Long term goal

GodspeedDiscvry, 2018-03-08:
Oh, my.... yep.

GodspeedDiscvry, 2018-05-03:
So, for all intensive purposes this is done for all currently active buttons and will be rolled in as each new button gets added.  The one exception is the back button.  We have hat drawn as a depress button - but it always floats on the black screen.  Do we still want to depress it?  Currently it shows the blue square while pressing.

GodspeedDiscvry, 2023-06-07:
Done years ago... and then some!

closed 2023-06-07 by GodspeedDiscvry