ISS-Mimic/Mimic #37: Pi setup issue?  Step 6 (Lightstreamer)

juggler2000, 2018-01-26:
Step 6) in the Pi Setup  looks like this:

[image]

But that seems like a "cd" to nowhere.  Issuing the commands as-is, and a few alternates, throw the errors in the screenshot below.

[image]

Ideas?

I was doing the install over VNC, but don't think there were any copy-paste syntax problems.

streadgold, 2018-01-26:
hmm yeah that is a blank cd just to make sure you're in the home directory, might be unnecessary.

Did step 4 complete successfully? Does "node -v" and "npm -v" return anything?

juggler2000, 2018-01-26:
User error -- missed step 4 somehow.

closed 2018-01-26 by juggler2000

juggler2000, 2018-01-26:
Hmm... I'm getting lots of colorful text from step &.  Perhaps I hosed something by doing steps 5 & 6 before 4.

[image]

streadgold, 2018-01-26:
Does it just keep looping that over and over? Try the other command in step 7 if so

juggler2000, 2018-01-26:
Seems to be having a downoad error... or something.

[image]

streadgold, 2018-01-26:
yeah that's the same problem susan was having. thought we solved it with that current setup list

juggler2000, 2018-01-26:
seems like it might be an amazon server issue.  I rebooted the pi and tried again, but same result.

juggler2000, 2018-01-26:
For step 12), if we are going to subject ppl to vi, we should prob provide the append and save&quit commands.

juggler2000, 2018-01-28:
kept trying the same things, and eventually worked!  GUI up and running! Live data!

streadgold, 2018-01-28:
nice! anything specific? perhaps you should save your command history so we can improve the setup steps if (when) others encounter that issue