ISS-Mimic/Mimic #38: geopy probs

juggler2000, 2018-02-04:
Maybe I picked a bad time to git pull.

[image]

streadgold, 2018-02-04:
oops forgot to add to the install instructions, this should resolve:

sudo pip install geopy

juggler2000, 2018-02-04:
Cool.  I was gunna  try something based on googles, but was worried that it
mattered where I installed from.  Does it?

streadgold, 2018-02-04:
Actually dont do that just re-pull the code. I'm switching from geopy cause it causes lag

juggler2000, 2018-02-04:
back in bidness! Thx!

juggler2000, 2018-02-19:
Hmmm....

Wouldn't let me pull, wanted me to stash my telem...db (old issue, right?).  So, I deleted that and pulled fine.

Running python GUI.Py then threw a traceback error:
--------
File "GUI.py", line 8 in <module>
      import reverse_geocode
ImportError: No module named reverse geocode.

ISS-Mimic, 2018-02-19:
oops, forgot to mention.

sudo pip install reverse-geocode

juggler2000, 2018-02-19:
Hmm... Some scipy complaints during the install. Will try more tomorrow.

closed 2018-02-27 by ISS-Mimic