ISS-Mimic/Mimic #57: 2B, 4B BGA sign issue

juggler2000, 2018-09-06:
2B and 4B signs appear flipped when mimic'ing ISS stream.  

This was initially confounding, because without the streaming data, they all seem to properly move to the startup position (30 deg).  But this is because the initialization values had signs flipped for these (and forgotten).
[image]

Likely need motor/encoder polarities swapped at the motor shield (or signs swapped in code).

ISS-Mimic, 2018-09-28:
@juggler2000 is this one solved? Or was this still the encoder problem?

juggler2000, 2018-09-29:
Solved, meaning I didn't change anything and it has been working fine.  There may be a phantom encoder problem that periodically manifests, so I'll take out insanity insurance.

closed 2018-09-29 by juggler2000