ISS-Mimic/Mimic #63: Extrusion Components

Digisembler, 2018-09-20:
8020 nut and cone for attaching to extrusions.
1/8 inch hole for rods
need design, stls and makerbot pla sliced files

Digisembler, 2018-09-26:
- [ ] Print SAW end caps (2)

- [ ] Print SAW canisters (2)

- [ ] Print Node 2 body

- [ ] Print Node 2 cap

- [ ] Print JEM PM

- [ ] Print JEM EF

- [ ] Print JEM ES

- [ ] Print replacement COF body

Digisembler, 2018-09-28:
- [ ] Explore ELC models from Blender model.

- [ ] Modify ELC1 models to have small magnet at CAS interface, FRAM/direct mount sites/ExPCA/Grapples
- [ ] Modify ELC2 models to have small magnet at CAS interface, FRAM/direct mount sites/ExPCA/Grapples
- [ ] Modify ELC3 models to have small magnet at CAS interface, FRAM/direct mount sites/ExPCA/Grapples
- [ ] Modify ELC4 models to have small magnet at CAS interface, FRAM/direct mount sites/ExPCA/Grapples
- [ ] Modify AMS model to have small magnet at CAS interface, Grapples

juggler2000, 2018-09-29:
This list is fantastic.  Which are print-ready?

Digisembler, 2018-09-30:
working... btw, i completed a Lab mod (and printed it) that has two beautiful endcones and a cylinder (PLA is my BFF!)
Also, need some help (thoughts/technical?) on how to setup/upload stls here...

streadgold, 2018-09-30:
If you go to the Code section, I just added a "Printing" folder, if you click on that then "Upload Files" at the top you can upload them there. Not sure what our file size limit is for our whole repository but STLs usually aren't that big right?

Digisembler, 2018-09-30:
Great thx.  Any way to edit the text with the commit after committing? i forgot to add the ELM PS print notes... (i.e. I'd add" Print the non-cbm end down, no supports. Again, PLA recommended.  Also setup for a magnet for the PIU interface on the top, same size as PM").

streadgold, 2018-10-01:
Not sure if you can edit a commit message

streadgold, 2018-10-01:
Oh so you can add comments to each commit, might be easier than editing it. 

Also Craig, your commit messages put mine to shame haha. Mine are all "added stuff, changed things"

Digisembler, 2019-01-03:
Ok, 2019!  I have the S6 (P6) array mount printed.
I fixed the S1 model (rotated the motor mount hole) and am printing now (~ 4 hrs)
We should clean up the above list of things to print, i think it's outdated and start a new one soon.

ISS-Mimic, 2019-01-23:
Yep, concur. Closing this issue, add any new components and tasks to Issue #77

closed 2019-01-23 by ISS-Mimic