ISS-Mimic/Mimic #66: Simulate orbit -> start yields crash

juggler2000, 2018-09-29:
Says "UnboundLocalError: Local variable 'beta1a2' referenced before assignment.  Live telemetry seems to work fine, and not cause this.

High likelihood of user error (need to press other buttons first?) but dunno.

have been "git pull"-ing every time before start, but I know some things on my Pi load are different.  For instance, I have to type "python GUI.py" rather than "mimic" to get things started.

streadgold, 2018-09-30:
Confirmed, same problem on my end. This is probably because for some reason I put the demo orbit stuff under mimic screen cause I totally forgot we actually made a screen just for that...

streadgold, 2018-09-30:
Solved (tentatively, needs to be tested with the actual motors of course). I moved all of the fake BGA stuff to the simulator orbit screen where it was supposed to go. Now pressing start demo will replace the sqlite database values with our recorded orbit and update them at 36x actual speed

closed 2018-09-30 by streadgold