ISS-Mimic/Mimic #67: P3 CAD model cavity not quite large enough to insert SARJ motor

juggler2000, 2018-09-29:
Cavity is perfect to house motor assembly, but not quite "long" enough to insert it due to shaft protrusion.  Could either elongate rectangular slot or make motor shaft hole a bit "deeper" towards centerline of truss.  Plan to file the one we have.

Digisembler, 2018-09-30:
ok. i agree for now to mod it - this is the same file as the blue one that works... is it possible that the motors have a bit of individual tolerance buildup? (yes, it is possible...)  can amend as needed.

Digisembler, 2018-10-04:
Dremeled slight angle to the axle hole, that did the trick.  Highly approve the instantaneous SLA support removal method (gravity assisted force applied to the object...) - but i don't suggest trying this at home. :P

closed 2018-10-04 by Digisembler