ISS-Mimic/Mimic #77: Transition model to new black aluminum structure

streadgold, 2018-12-03:
- [ ] Requires new mounting interface for S3/P3

- [ ] Might require other changes for module stack support?

Digisembler, 2019-01-24:
Modified the truss endcaps to match the new heights.  Printed those in black PLA.
Reprinting the S1/P1 Mounts in black PLA (fixes/completes S3/P3 new mounting above).
#3 x 36" threadrod and nuts ordered to tighten inboard truss.  (really wondering why we didn't just use a flat bar and set screws in each module... would have been easier/tighter... just thought of it, so, that's probably why, but doesn't really keep the stack axially tightened.)

ISS-Mimic, 2019-01-24:
Bueno! If it was difficult to find the rod/nuts it might be worthwhile to link the part numbers / links to the products somewhere here for future reference

Digisembler, 2019-01-24:
Nuts:
https://www.grainger.com/product/2GA49?cm_mmc=EMT-_-10376092-_-ShippingConf2018NavBar-_-OnlineGuest-KPIRPE-_-Item&RIID=50635142275&GID=&mid=EMT-_-10376092-_-ShipConfirmation-_-Item&rfe=6b8e4fd0d0254c6f348a2870942a7662d7f525f21e1fba5d2722ed2880f81439&gcrfe=6b8e4fd0d0254c6f348a2870942a7662d7f525f21e1fba5d2722ed2880f81439

Threadrod:
https://www.grainger.com/product/21YN98?cm_mmc=EMT-_-10339122-_-OrderConf2018NavBar-_-OnlineGuest-KPIRPE-_-Item&RIID=50635142275&GID=&mid=EMT-_-10339122-_-OrderConfirmation-_-Item&rfe=6b8e4fd0d0254c6f348a2870942a7662d7f525f21e1fba5d2722ed2880f81439&gcrfe=6b8e4fd0d0254c6f348a2870942a7662d7f525f21e1fba5d2722ed2880f81439

We'll see if these work when they get in.

Digisembler, 2019-02-06:
Threadrods delivered, cut, installed with nuts.  New zip-tie outboard support blocks designed/printed (need small zip ties for today... no wider than 1/8", probably 3-5" long, minimum of 4 needed)

ISS-Mimic, 2019-02-07:
Completed - or at least back to where we were on the old frame. Will bring zip ties tomorrow

closed 2019-02-07 by ISS-Mimic