ISS-Mimic/Mimic #80: Truss Cable Choice

streadgold, 2018-12-19:
Related to Issue #74 

We talked about using cat-7 cables for the runs down from the truss back to the arduino but maybe we should actually use USB-C?

Pros:
USB-C are 12-pin cables, perfect for our 12-pin slip rings. So 4 cables total coming from truss, instead of 8. 
USB-C are slimmer and look more high-tech?

Cons:
Breakout board would have slip ring wires soldered directly on, wouldn't look as clean.
Breakout board is a little wide and wouldn't hide as well as ethernet doubler 

Thoughts?
@Digisembler @GodspeedDiscvry @juggler2000

streadgold, 2019-01-07:
USB-C won't work :( 

They require magic electrical nonsense to work. Back to ethernet

ISS-Mimic, 2019-02-06:
Ethernet it is.

closed 2019-02-06 by ISS-Mimic