ISS-Mimic/Mimic #97: Zero Command

ISS-Mimic, 2019-02-20:
Calibrate screen will send a command to the arduino to instruct it to "zero" all the joints. Thinking command will be "ZERO". Upon this command, every joint angle should think it's current position is zero.

juggler2000, 2019-02-21:
We need this for 2x SARJs and 8x BGAs.  We do not need this for TRRJs.

There are two ways to approach this:
1) RasPi sends the following commands:
Count_B2B = 0
Count_B4B =0
Count_B2A = 0
Count_B4A =0
Count_B1B = 0
Count_B3B =0
Count_B1A = 0
Count_B3A =0
Count_SSARJ=0
Count_PSARJ=0

2) RasPi sends "Zero" command, and the Ard assigns all of the above.

juggler2000, 2019-02-25:
Thinking about this more, I think it makes sense to do it from the Arduino side.  We'll have to set all of the "Count_*" and "*cmd" vals to zero at the same time to avoid it suddenly jumping.