ISS-Mimic/Mimic #100: Save ISS Live

streadgold, 2019-03-06:
Need to determine state of ISS Live and if necessary save it

streadgold, 2019-04-09:
It has now been confirmed by NASA that ISS live was defunded and there are no efforts to revive it :(

ISS-Mimic, 2019-04-26:
:warning: Not to get overly dramatic but the ENTIRE FATE OF ISSLIVE AND MIMIC COULD BE DECIDED THIS COMING TUESDAY :warning: :warning: :warning: :satellite: :satellite: :satellite: :earth_americas: :rocket: :rocket: :stars: :star2:

ISS-Mimic, 2019-08-30:
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!WE DID IT!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! :tada: :confetti_ball: :rocket: :confetti_ball: :tada: Thanks mostly to Bryan's extraordinary ability to pester the right people, ISSlive will be returning to an interweb near you this fall!

Not closing the issue until it is back up and running but this is really happening and it is mostly because of this project. Epic epic epic epic epic.

:tada: :confetti_ball: :rocket: :confetti_ball: :tada: :tada: :confetti_ball: :rocket: :confetti_ball: :tada: 
:tada: :confetti_ball: :rocket: :confetti_ball: :tada: :tada: :confetti_ball: :rocket: :confetti_ball: :tada: 
:tada: :confetti_ball: :rocket: :confetti_ball: :tada: :tada: :confetti_ball: :rocket: :confetti_ball: :tada: 
:tada: :confetti_ball: :rocket: :confetti_ball: :tada: :tada: :confetti_ball: :rocket: :confetti_ball: :tada:

streadgold, 2019-09-16:
ISSlive! is officially back up and running

http://demos.lightstreamer.com/ISSLive/
isslive.com

ISS-Mimic, 2019-09-16:
CLOSING ISSUE!!!!!!!!!!!

closed 2019-09-16 by ISS-Mimic

dziban303, 2021-01-12:
That didn't last very long. It's been down since June

ISS-Mimic, 2021-01-12:
> That didn't last very long. It's been down since June 2020.

Hi dziban303, that's not quite accurate. We record all the telemetry coming off ISSlive and monitor it's status constantly.

ISSlive has been offline since December 15th due to the Solarwinds cybersecurity incident, however the system is scheduled to be restored sometime this week. This current outage is the longest the system has been offline since it was restored back in September 2019.

dziban303, 2021-01-12:
Thanks for the reply! I last checked it when the most recent hunt for an air leak was going on, so I guess that was September, and it was working then; when looking for some news about why it was down now, I saw tweets from people saying it was down going back to June. After hearing about the plan to install new solar panels next year, I checked this evening to see how much power they were producing now. 

Anyway I'm glad it'll be back up soon. Having the ability to see what's going on with ISS makes it much more "real." 

Cheers

ISS-Mimic, 2021-01-12:
No problem! There are occasionally a few days or a week long outages due to server issues but if it lasts longer than a week we'll reach out and make sure the team is aware that the public data feed isn't working. 

I'll need to graph the US lab cabin pressures from the air leak sometime, should have that data recorded. 

We're excited about the new solar arrays too, from the public telemetry (if I recall correctly) the solar array current/voltage values actually reflect the DCSU output, so it's more like the actual downstream power usage than power production capability. But that value seems to typically be around 40-60kW. Will be interesting to see how things change with the new arrays

ISS-Mimic, 2021-01-20:
Heard back from ISSlive that they are still having problems but definitely still working on restoring the public data stream

ISS-Mimic, 2021-01-22:
The telemetry has returned!

dziban303, 2021-01-22:
So it is! Thanks for the update. https://isslive.com is still not working. I'm actually glad it broke, because otherwise I wouldn't have found this project, which I think is super cool. Cheers!

ISS-Mimic, 2021-01-22:
Not sure what the problem with isslive.com is, but that site is no longer run by NASA so not sure how much effort is being made to maintain it. 

Glad you found us though! We have our two pages that show all the live telemetry https://iss-mimic.github.io/Mimic/ and https://iss-mimic.github.io/Mimic/dashboard.html (and fun fact about those pages, we might be the only place on the web to view ALL OF THE TELEMETRY! that I know of at least, because the Canadian Space Agency got a bunch of robotics data added after the launch of isslive and to my knowledge no one else is displaying that data).

While our two web pages aren't as graphically interesting and isslive.com, our Raspberry Pi software, while still definitely in development, has the goal to display all of the telemetry in a more interesting graphical format than isslive.com.

The Raspi GUI currently shows a few things in its current state like the solar array data with animations showing whether or not the arrays are sunlit, and if the batteries are charging/discharing. Also has a needle gauge that shows the pressure in the airlock which is fun to watch during spacewalk prep, and we also show which satellite the SGANT antenna is pointing to and which antennas are broadcasting. We are always looking for new ideas on interesting ways to visualize the telemetry without just showing the raw numbers so let us know if you have any ideas. I am also putting together a wishlist of telemetry that is not currently available in the public stream but that I plan to request they add to the public telemetry stream.

The full list of public telemetry with detailed info can be seen here by the way: https://github.com/ISS-Mimic/Mimic/blob/master/ISS_Public_Telemetry.xlsx