ISS-Mimic/Mimic #118: Potential SARJ drift

ISS-Mimic, 2019-10-14:
When running in live mode for several hours it looks like the SARJs begin to drift by quite a bit. Both stay in sync with each other but they both appeared to be roughly 45 degrees off the actual angle after running for approximately 4 hours.

SWGlassPit, 2019-10-14:
Does the drift happen if the joints are commanded through multiple full
rotations via the calibration screen? Might rule out something simple like
having the wrong number of encoder pulses per rotation

ISS-Mimic, 2019-10-14:
Yes drift still occurs through manual control, a single full rotation appears to be about 30 degrees off

ISS-Mimic, 2019-10-24:
@juggler2000 do we think this is solved now?

juggler2000, 2019-10-24:
Yes, solved.

closed 2019-10-24 by ISS-Mimic