ISS-Mimic/Mimic #124: Major power draw problem

ISS-Mimic, 2019-10-28:
Some kind of issue where we seem to be drawing too much current from the Raspberry Pi USB ports, likely due to the added neopixels and servo motors being powered directly off the USB port. 

Seems like the solution is to separately power the servos and/or the neopixels so they don't all draw from the Raspi USB ports which should not be used to provide more than 1.2A total from all 4 ports.

juggler2000, 2019-10-30:
Hmmm... disconnecting one of the servos solves the problem.  Sometimes, so does disconnecting half of the LEDs.  Initial test of powering STBD servo separately (from hacked USB cable plugged into same battery power brick that's powering the Pi). Yielded good results, but only from nominal motor commanding, so nothing drawing much current.

streadgold, 2020-11-14:
Solved with the powered USB hub

closed 2020-11-14 by streadgold