ISS-Mimic/Mimic #130: Pi needs serial monitors

juggler2000, 2019-11-16:
Need to have serial monitors for each connected device, since the motor-driving Ards send info back.

This would be Uber helpful when debugging motor oddness.

streadgold, 2019-11-18:
So reading from the serial ports shouldn't be hard. Where would we display the data from the arduinos? Presumably most of the debugging would be done from the manual control screen, which is pretty busy. We could have it in the status bar at the bottom but that is only good for one ish line at a time. Or have a subscreen that has a lot more real estate but you'd have to switch back and forth to issue commands. Thoughts?

streadgold, 2020-11-13:
@juggler2000 given all the serial difficulties I'm not sure this will ever happen - is this issue also solved by just using the LCD screens on the HDMI shields?

juggler2000, 2020-11-13:
Yep, LCD screens should be just as good.  Maybe better, since we could always have data scrolling.  The downside is that we won't be able to see all of the data.

streadgold, 2020-11-14:
Closing for now since this likely won't happen. If serial usage works out well with the new code we may revisit this later

closed 2020-11-14 by streadgold