ISS-Mimic/Mimic #134: SAW stiffening

juggler2000, 2019-11-16:
Perhaps an alternate SAW build.

Some thoughts:
- blankets need to be more rigid, for transport and durability
- All/mostly printable version would be nice (single piece would be too large for most printers)
-

GodspeedDiscvry, 2019-12-04:
Just for documentation purposes because everyone seems to keep forgetting.  The current soft SAWs needed to be redone due to a miscalculation in measurement - that is why they are loose and fall apart during translation.  There are also several that are broken now - both situations would be remedied with new prints.  It has not been done because printer was not working/or has not been around lately.  They would be much more stable and rigid if we could get new prints.

ISS-Mimic, 2020-01-23:
I think we can close this one. Still some fine tuning to be worked out with the 3D printed arrays but they definitely seem to solve the stiffness/reliability issues

closed 2020-01-23 by ISS-Mimic