ISS-Mimic/Mimic #137: Information needed: Telemetry use for real-time simulation in SpaceCRAFT VR Platform

connorjak, 2019-12-04:
Hello,

I am Connor Jakubik, System Architecture Team Lead at SpaceCRAFT VR at Texas A&M University. We are putting together a new, open-source integrated simulation platform that is releasing somewhat soon. For one of our example simulations, we would like to have real-time ISS telemetry feeding the state of our ISS model. In particular, we would like to at least have the positions of the solar panels and radiators, but the more parameters the merrier.

Upon glancing over this codebase, it looks like the majority of this code is Javascript and Python. While we are working on eventual Python support, we use C++ for the majority of our code. How should we proceed with trying to get this data stream to C++? One idea that comes to mind is using a file / named pipe for communicating events.

connorjak, 2019-12-04:
We would definitely contribute our C++ interface back to this repo, and we might be able to help with your goals as far as a digital 3D representation.

connorjak, 2019-12-04:
I started working on this, but it appears either I or the maintainer of the cpp-lsclient (c++ lightstreamer client) library have done something wrong. (https://github.com/quotekio/cpp-lsclient). I see nothing but the initial HTTP request to the server. I can post my code so far if someone would like to take a look.

ISS-Mimic, 2019-12-04:
Hi Conner, that sounds like an awesome project and we'd love to help out! 

We'll look into a c++ interface and see what it takes, but I wonder if you can do exactly what we did to solve a similar issue when trying to get this data stream to work in python? When we started there was no python client for lightstreamer so we are just using the javascript code to constantly dump all of the telemetry into an sqlite database that is simultaneously being read into the main python code, would that work for your application but instead using c++ to access the database?

SWGlassPit, 2019-12-04:
Gonna chime in here to agree. The current concept uses the SQLite database
to provide a clean break between the code that receives the telemetry and
the code that uses the telemetry. In fact, we have an unreleased bit of
code designed to play back recorded telemetry over the same interface. It's
written in C++, and it's a simple drop-in replacement for the JavaScript
that receives the telemetry from the server. For the Python code that uses
the telemetry, there is no apparent difference.

For your needs, I would recommend leaving the JavaScript as is (maybe with
modification for where the database is that it writes to) and using
whatever your language of best comfort is to read and use the telemetry
from that db file.

connorjak, 2019-12-05:
Thanks, yeah that sounds like a good solution. I should be able to start the Javascript from my C++ code. Where is the Python code you're talking about that accesses the SQLite database?

ISS-Mimic, 2019-12-05:
This file : [database_initialize.py](https://github.com/ISS-Mimic/Mimic/blob/master/Pi/database_initialize.py) initializes the blank database (to be filled by the javascript once started) and then our main code: [GUI.py](https://github.com/ISS-Mimic/Mimic/blob/master/Pi/GUI.py) 

Connects to the database:
lines 101-103
`conn = sqlite3.connect('/dev/shm/iss_telemetry.db')`
`conn.isolation_level = None`
`c = conn.cursor()`

And then retrieves the telemetry, checking for updates every second:
lines 2152-2160
`c.execute('select Value from telemetry')`
`values = c.fetchall()`
`c.execute('select Timestamp from telemetry')`
`timestamps = c.fetchall()`

`sub_status = str((values[255])[0]) #lightstreamer subscript checker`
`client_status = str((values[256])[0]) #lightstreamer client checker`

`psarj = "{:.2f}".format(float((values[0])[0]))`
...

connorjak, 2019-12-05:
It's getting close to working, just having a little trouble with the specifics of the sqlite3 C API. The occasional LOS of station data isn't helping either.

connorjak, 2019-12-05:
Everything seems to be working besides the SQLite3 select queries in the C++ code. I'm getting good data on the Javascript side, but when my C++ SQLite queries execute they time out after 10 seconds with a Locking Protocol error (code 15 https://www.sqlite.org/rescode.html). I'm going to make a StackOverflow question for this.

connorjak, 2019-12-05:
https://stackoverflow.com/questions/59192840/isslive-telemetry-locking-protocol-sqlite3-c-error-on-query-with-javascript

SWGlassPit, 2019-12-05:
Have you verified that the database is operating with WAL journal mode? It
requires at least SQLite version 3.7.0.  That's necessary to allow
concurrent reads and writes without the use of locks.

ISS-Mimic, 2019-12-05:
I forgot to mention also that we delete the database file and create it again from scratch every single time the program runs to solve what I believe was a similar problem. That's why we have that database initialize script which gets called on program startup.

"The occasional LOS of station data isn't helping either." - You have no idea how often we say this :P

SWGlassPit, 2019-12-05:
To further this: you'll notice that the code we use places the database in
/dev/shm, i.e., it doesn't even exist in permanent storage.

Data dropouts are definitely a thing. There is generally a brief one
(usually less than a minute) about every thirty minutes as the ISS changes
which TDRS satellite it relays data through. This is because directional
antennas have to physically move and point at the new satellite.

From time to time, there will be longer dropouts as ISS passes over
longitudes that include central Asia, India, and the Indian ocean. That's
because the TDRS satellite in that location is not always available to ISS
(ISS is the number 2 priority customer of the TDRS system. The number 1
priority customer is not publicly identified). On maps projected on the
mission control screens, the region where this happens is identified as
"ZOE", the zone of exclusion.

Then there are even longer dropouts that we just don't have any explanation
for. Flight directors have the ability to kill the public stream, but it's
not clear that this has ever been exercised. It's entirely possible these
may be just due to a problem with the data adapter that feeds the
lightstreamer server.

Then there was that dropout that lasted roughly from February to September.
That was because NASA decided to shut it down. It took a decent number of
people with quite a bit of effort to convince them to walk that back.

connorjak, 2019-12-06:
> I forgot to mention also that we delete the database file and create it again from scratch every single time the program runs to solve what I believe was a similar problem. That's why we have that database initialize script which gets called on program startup.

I do that database_initialize.py script every time I start my C++ code. My c++ code is as close as I can get it to your GUI.py in order of operations.

SWGlassPit, 2019-12-06:
I highly recommend reimplementing the database_initialize.py script
natively in c++ rather than using system() to invoke it. That will have as
number of benefits:

1. It eliminates an unnecessary dependency on Python that you wouldn't be
using otherwise (that script does nothing python-ish. It's just a bunch of
SQL calls)

2. It eliminates the use of the unsafe system() call

3. It allows better capturing of SQLite errors.

The lock errors suggest that the `pragma journal_mode=wal` line is failing.
Can you verify your certain if SQLite is at least 3.7.0?

connorjak, 2019-12-06:
I reimplemented the database_initialize.py script in C++ and have had mixed results. The code has worked at least three times (c++ receives and prints the real values), but a common case I get is having every value and timestamp be zeros, and Javascript giving me a Locking Protocol error. I have sqlite3 version 3.22.0 in my C++ code. Checking javascript...

ISS-Mimic, 2019-12-06:
If the WAL journal mode was added in 3.7.0 you might want to update sqlite version and see if that help since that code 15 error specifically mentions WAL mode

connorjak, 2019-12-06:
But, 3.22 > 3.7

connorjak, 2019-12-06:
Javascript package is version 4.1.1 of sqlite3. Could it be possible that this version difference is what is breaking the code? I could try to get a more up-to-date sqlite3 C library, or possibly an older JS one.

ISS-Mimic, 2019-12-06:
Hah oops good point didn't read correctly. I remember us getting the locked database error tons of times at the beginning but our code as is never has that problem anymore. If you run the javascript on it's own does it correctly populate the database without issue? I also remember have a lot of problems with the javascript sqlite package

connorjak, 2019-12-06:
Oh nvm that was a misnomer; here's node-sqlite3's changelog (still on sqlite3 3.x)

connorjak, 2019-12-06:
Here's a public gist of my code: https://gist.github.com/connorjak/9f853bb83559976e12c9e38811c3c0af

connorjak, 2019-12-06:
To confirm, the exact order of your code is: 
1. Python `database_initialize.py` removes then reinitializes a SQLite3 database in `/dev/shm/iss_telemetry.db`
2. Python `GUI.py` establishes a SQLite3 connection to the same directory, applies `isolation_level = None` to the connection object
3. At some point in `GUI.py`, a Python subprocess is opened for `ISS_Telemetry.js`
4. NodeJS `ISS_Telemetry.js` establishes a SQLite3 connection to the same directory, with this line: ```var db = new sqlite3.Database("/dev/shm/iss_telemetry.db", sqlite3.OPEN_CREATE | sqlite3.OPEN_READWRITE);```
5. `ISS_Telemetry.js` establishes a LightstreamerClient connection to the lightstreamer server, with `setSlowingEnabled(false)`
6. `ISS_Telemetry.js` listens on the Lightstreamer connection and updates the SQLite database with new data.
7. `GUI.py` goes on to repeatedly execute select queries and fetchall() to get the entirety of the telemetry every tick.

(ISS_Telemetry.js and GUI.py run at the same time, with GUI.py starting ISS_Telemetry.js)

connorjak, 2019-12-06:
The javascript on its own reads the lightstreamer data correctly. I don't have code at the moment that can tell if the database is being populated by it correctly.

ISS-Mimic, 2019-12-06:
Yes that is precisely the correct code order. Also I just ran your linked javascript code and can confirm it seems to be updating the database correctly. I'm out of ideas until Tristan provides some more insight

He wrote a c++ program that takes in recorded data and updates the database as though it were live, that code might be useful to you [mimic-fakedata.cpp](https://github.com/ISS-Mimic/Mimic/blob/master/Pi/RecordedData/mimic-fakedata.cpp)

connorjak, 2019-12-06:
I updated the gist. I think for now, I'll try some options besides the SQLite database, like a socket connection between NodeJS and C++.

connorjak, 2019-12-06:
I got it working! I modified the Javascript to use a net socket, which I connected to in C++ and parsed serialized JSON "by hand". There is TONS of room for improvement, but it WORKS!

connorjak, 2019-12-06:
(Reopened for discussion of solution)

ISS-Mimic, 2019-12-06:
Excellent! I still wish we could figure out the sqlite route problems too but that's a great solution

SWGlassPit, 2019-12-06:
Glad you got it working. If you are at all interested in eliminating the
JavaScript dependency in the future, lightstreamer (after we already
developed the JavaScript receiver) released a more general, curl-based api
that allows you to perform the same task without being nailed down to any
particular language. As a proof of concept (that sadly isn't in the repo at
all), I did get it working in plain bash on a Linux terminal window, so
that's an option to consider for the future

connorjak, 2019-12-07:
Thanks to y'all for the help. Here's some video from SpaceCRAFT (real-time, solar panels and radiators being moved by telemetry): https://drive.google.com/file/d/1MGMc6ADWElYGD8hQ-CqmbrH7b5ajTBei/view?usp=sharing

SWGlassPit, 2019-12-07:
That is some incredibly impressive work there!

streadgold, 2019-12-07:
That's amazing! What source material did you use to create the model?

connorjak, 2019-12-07:
I think we got most of the 3D model components and mapped textures online for free or low cost; I can ask. There were quite a few model fixes we had to make though. I personally did some of the ISS interior and exterior lighting and materials, and I did the super-high-detail Earth we have (besides the atmospheric scattering; that’s a plugin).

streadgold, 2019-12-07:
That is truly amazing, great work! You might want to move PMA3 to Node 2 Zenith though :P then it will be perfect. Got any videos of the interior? I'm completely astounded by how great that model looks though really amazing

streadgold, 2019-12-08:
Ah wasn't loading on mobile I see the interior now very cool. In the telemetry stream there's also data for the MT position and the SSRMS angles, which would be an extremely cool addition if you're looking for anything else to add. Would also be really cool to watch live during robotics ops

connorjak, 2019-12-08:
My ideas for stuff in the telemetry that would be especially cool to add:

- SPDS/SSRMS (Canadarm) movement (Rather hard because we might need to wait for a berthing / EVA to test)
- LOS/AOS notifications in-sim
- Power production visualization
- Display/Visualize relevant parameters in various places on the interior
- TDRS / GPS comms

Do y'all have recorded telemetry data that includes SSRMS movement? It's rather hard to test that when you have to wait for an EVA or berthing.

connorjak, 2019-12-08:
I updated ISS_Telemetry_IPC.js with some text coloring in the terminal output. Not sure if that helps you guys, but it aids in picking out important info in our (crowded) terminal output.

ISS-Mimic, 2019-12-08:
"It's rather hard to test that when you have to wait for an EVA or berthing." Hah!! yep we're currently waiting for an MT translation to test an animation. And yep we do have recorded telemetry with SSRMS + SPDM + MT movement. Here ya go [telemetry_2019-09-30_23_59](https://drive.google.com/open?id=1n3z85oyhQEDm_hRS0CudAv71QC_N0l2f) and [telemetry_2019-10-01_23_59](https://drive.google.com/open?id=1KxsyPNMGA9kDeC8PKMl9Td-uRzt23KzD)

Just tried your text coloring js, is pretty cool I didn't even know you could do that haha. 

Let us know if you want recorded telemetry for any other events, we are recording everything constantly. 

Also we're putting together a wish list of telemetry we're going to request be added to the public stream so if you have anything else that would be useful feel free to suggest it.

connorjak, 2019-12-09:
Thanks for the recordings!

Additional Telemetry values I would like (some of which might be hard to get due to security concerns):
- Exterior lights being on or off / brightness value
- Interior lights / other conditions that would show up visually?
- Communication equipment activity
  - RF equipment pointing data
  - Activity / power level / data rates / recipients
- Anything about live video feeds:
  - For example, during proximity/robotic/EVA operations, getting camera pan/tilt/zoom/exposure to be able to match live ISS video with a digital reproduction
- Visiting vehicle telemetry, to be able to recreate dockings/berthings
 - I think there's a little bit of this in the Russian telemetry; KURS rangefinder stuff IIRC
- Sensor data that could assist in graphics/rendering validation, especially for brightnesses in certain directions

I'm not sure what is available, but we would like pretty much anything we can get. One of our research goals with our Platform is to evaluate VR's ability to improve human performance in extreme environments, and a perfect candidate for that is the ISS. Anything that can either validate our graphics, propagation and representation of the Station is great for us. An especially interesting operation for us is robotic ops with an astronaut on the SSRMS.

SWGlassPit, 2019-12-09:
The good news is that some of this information is already available. The RF
pointing and activity is present in the items referring to the SASA
(steerable horn antennas on S1 and P1, telemetry includes azimuth and
elevation angles - SASA connectivity is a necessary condition to receive
telemetry at all) and the SGANT (steerable dishes on Z1 - elevation and
cross-elevation, along with transmitting status)

ISS-Mimic, 2019-12-09:
Those are great suggestions, yep as swglasspit noted RF pointing data is available for all gimbal-ed antennas (with the exception that we currently can't tell which SGANT is active - but that is on my list). I have power level and recipients (VV ID) on my list as well. KURS range and status is currently available. 

Visiting vehicle stuff is definitely not going to be approved but the rest of it I'll definitely add to the list. Thanks for the suggestions

connorjak, 2019-12-09:
Oh, looks like I hadn't dug deep enough, that's good that we have the RF pointing + activity.

Another question: do we have a way of determining exact latency between the feed and current time? Right now I am experiencing differences from sim to NASA livestream of about 30 seconds, and I'm not entirely sure where it's coming from. The livestream is 30 seconds ahead of my sim with the telemetry.

SWGlassPit, 2019-12-09:
When we took the mimic project into the Mission Evaluation Room in NASA's
mission control building, it appeared the delay between real time and the
public telemetry feed was somewhere around 7 or 8 seconds.

connorjak, 2019-12-09:
I see. I probably have some error or delay in:
1. My code that gets the current UTC time.
2. My code that converts that into seconds from J2000 epoch.
3. My code that feeds that into NAIF SPICE for planet/moon positions/rotations.
4. NAIF SPICE (unlikely).
5. My Earth model’s exact rotation (before being rotated by SPICE.
6. The position and velocity I choose as my initial state for an orbit propagator for the ISS, which I get from the telemetry, but don’t check the time stamp on.
7. A rift in the space-time continuum allowing the video livestream to arrive to my computer before the corresponding telemetry has.

Come to think of it, I haven’t really verified 6 here. The state vector telemetry isn’t updated very often. I could probably get a better fix on the station position by either setting initial state only upon the reception of a fresh state vector, or doing something fancy with the timestamp.

There also might be a bit of time weirdness (leap seconds, computer clock being slightly out of sync, etc) going on as well with the conversion to seconds past J2K. 

Do y’all know something about the time parameters in the telemetry? I didn’t recognize any of the numbers (seconds from J2K is about 6 million), but if I could modify my planets propagation to sync up with the ISS telemetry, that would also work.

SWGlassPit, 2019-12-09:
An initial attempt I made to convert the reported state vector to ECEF
coordinates and correlate it with an SGP4 based propagator (with TLEs
reported from other public sources) resulted in a consistent 5 or so degree
longitude lag, which appears to match your experience here. I haven't
explored further.

closed 2020-10-11 by streadgold