ISS-Mimic/Mimic #159: Current arduino code

ISS-Mimic, 2020-05-04:
https://github.com/ISS-Mimic/Mimic/blob/b3fa0ea01785515d7e5b14d8b56a029773309429/Arduino/Ard_PortBGAs_r15/Ard_PortBGAs_r15.ino#L9

Hey is this the most recent code for the BGA arduinos? I'm working on the HDMI shield and trying to map all the pins correctly and unambiguously. What do encoder pins 16,17 refer to?

juggler2000, 2020-05-05:
It's the r15's (not the Combo-loco)
https://github.com/ISS-Mimic/Mimic/tree/master/Arduino/Ard_StbdBGAs_r15
https://github.com/ISS-Mimic/Mimic/tree/master/Arduino/Ard_PortBGAs_r15

juggler2000, 2020-05-05:
Encoder pin numbers are in parenthesis for the code line excepts below.
Stbd: 
174 Encoder myEnc1A(0, 1); 
175 Encoder myEnc3A(2, 3); 
179 Encoder myEnc1B(7, 8); 
180 Encoder myEnc3B(11, 12); 

Port:
174 Encoder myEnc4A(0, 1); 
175 Encoder myEnc2A(2, 3); 
179 Encoder myEnc4B(7, 8); 
180 Encoder myEnc2B(11, 12);

closed 2020-08-08 by ISS-Mimic