ISS-Mimic/Mimic #170: Printed Microcontroller Holder

ISS-Mimic, 2020-08-31:
Would be nice to have a method of securing the Metro board to the frame.

CAD model of the metro should be here:

https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2488%20Adafruit%20Metro%20328

cactus314, 2021-06-29:
Preference for a holder that can be fastened to the 2040 extrusion or just held in place with a light press-fit?
[image]

Digisembler, 2021-07-21:
Patrick's solution seems to work well, might watch out for future issue of the screw hole pillars breaking off

cactus314, 2021-07-25:
Uploaded the Metro 2040 Mount to the structural components directory of the 3D Printing folder.  I'll let @streadgold rename/move it if he wants.  It fits the middle Metro pushed all the way against the t-brackets without too much side torque from the HDMI connection.  I'll design a separate mount for the Metro Mini I intend to use for the NeoPixels.
[image]

cactus314, 2021-07-28:
Uploaded STLs for mounts for the Metro Mini (i.e, Arduino Nano) considered for use to control Neopixels,  as well as a mount for the Sabrent 4-Port USB Hub.  These additions might close out this issue.  I'll let someone else (@streadgold / @ISS-Mimic ) decide.
[image]
[image]

streadgold, 2021-07-28:
Nice! Looks good to me

closed 2021-07-28 by streadgold