ISS-Mimic/Mimic #171: Include keyboard app to RasPi SD card image

juggler2000, 2020-09-02:
For convenience, include the keyboard app on the RasPi SD card image.  Also include "mimic" shortcut in cmd line, rather than requiring "python GUI.py" to execute.   Better yet, desktop icon to launch it.

juggler2000, 2021-01-27:
Would be nice to have the virtual keyboard and mimic shortcut included on an updated SD card image in time for SEEC.

dallasbrittany, 2021-01-27:
That's a good idea. I was doing that myself when I set things up anyway. I was holding off on making the new image though because of a package issue documented in a separate ticket that needs investigating.

dallasbrittany, 2021-01-27:
Well, I wasn't doing the desktop icon. That can have some unintended consequences. We don't have a real uninstall process because we aren't doing packages, and we would want to clean up icons if someone removed mimic. Also, what if you have more than one mimic directory? (I usually do.) And if we did want to delete the mimic icon, sometimes bad things happen if you delete it and there's nothing else on the desktop. Lubuntu has been known to get rid of the desktop folder entirely. I know we aren't using that distro, but still.

juggler2000, 2021-01-27:
My thought is that the SD card image is mostly for non-developers and can be assumed to include those new to RasPi.  So, my preference is to have everything installed and prepped for simplest usage.  

This would include the keyboard, desktop shortcut (if it doesn't introduce problems), and maybe another shortcut for a "git pull" update.

Keyboard seems like the minimum to include, since then they won't have to temporarily source a physical keyboard just to get the virtual one installed.

closed 2021-08-11 by streadgold