ISS-Mimic/Mimic #187: Replace LED-driving Ard Mega with Nano or similar

juggler2000, 2020-10-14:
Just to have smaller footprint on the frame.  Of course, we eventually want a single Ard running everything....

cactus314, 2021-08-17:
So far:
- Metro Mini is recognized by the RPi, but doesn't register serial commands from it.  (I've reached out for advice on the subject from a knowledgeable friend.)
- Trinket M0 registers the serial commands, but has only a single ground pin - need at least two
- Will try a Nano clone next.  Don't expect any problems, but Nanos have a Mini USB connector rather than the preferred Micro USB
- Have ordered a Sparkfun Pro Mini to try as well

cactus314, 2021-08-22:
- Still have no success with the Metro Mini, but will continue to try.  It would be the best solution if only it would work.
- The Nano clone seemed to work, at least as well as I could tell from the STEM model which has some 'finicky' LEDs, making it hard to validate the Nano as a robust solution.  There's still the issue of having to use a USB-Micro cable for the Nano, plus there are circuit elements mounted to the bottom of the board, making it a little more challenging to design a proper holder.
- The Sparkfun Pro Mini is like the Metro Mini - it is "seen" by the RPi, but the Pro Mini doesn't appear to register any serial commands sent to it by the RPi.  Also, from Sparkfun's own webpage: "_Getting the Pro Micro or Fio v3 set up on your computer and in your Arduino environment can be difficult._"  It really wasn't any worse than setting up the Arduino IDE for the Adafruit Metro 0's, but it's also easy to brick the Sparkfun Pro Micros - I almost did.  Not sure I like it as a solution.

cactus314, 2021-08-24:
I do have a Nano clone working to control the Neopixels, but it's a bit persnickety.  Sometimes you have to reset it for the serial data to register, then it shows up as a 5th Arduino connected to the RPi on the Mimic GUI.  Not sure what's causing that.  It does take up a smaller footprint (see comparison images below), but I can only find them in 3-packs without the headers soldered in place, so there's not much cost savings over just using a 4th Metro 0, except then you get two extra Nanos to play with.  The Metro 0 never seems to give any problems and simplifies uploading the firmware (only one board to have to set configurations for), and would mean all the USB cables would be the same (the Nano would require a USB-Mini).  I'm not sure the benefits are all that great.
I'll let y'all vote on whether to go forward with using a Nano or stick with a 4th Metro 0 and consider closing this ticket.  I'm fine with either solution.

[image]
[image]

ISS-Mimic, 2021-08-24:
Alright, I'm fine with just using a 4th metro then, seems like the best option. 

-Sam