ISS-Mimic/Mimic #192: Improve Serial Port Detection

ISS-Mimic, 2020-10-29:
Currently, the serial ports are hard coded to ACM0 ACM1 ACM2 and ACM3. These are the most comment ports for our devices, but other devices or more/less devices would not work.

Added code to the "serial_ports" which checks all possible ports - this worked great but when actually using it when the connected devices there was considerable lag. We call it every 1 second ish and try to write to all those ports so... yeah.

Would be nice to detect whenever a device is plugged in or removed and update arduino_count accordingly (also need to handle other serial ports)

dallasbrittany, 2020-11-08:
Are you okay with having to pip3 install pyudev? I can add it to the instructions list.

I changed things so you can either fill out config.json and then run with python3 GUI.py --config, or you can just run the normal way, and it'll find things with ID_VENDOR that includes Arduino, Adafruit, or FTDI. Metro M0 boards seem to use Adafruit, but I wanted to be able to support other Arduino boards too. I tested with the Uno (Arduino), Mega (Arduino), and Nano (FTDI).

I haven't made it so it detects when you plug something in. I figured it was good enough to just make sure you plug things in before you start the GUI. But I can make it detect if you think it's useful. What do you think? I'm not sure what your use case is like.

streadgold, 2020-11-08:
Sure! Asynchronous device monitoring sounds great

dallasbrittany, 2020-11-18:
Yeah, it would really help if I could type correctly the first time.

This:
`NameError: name 'OPEN_SERIAL_PORT' is not defined`
happened because it's supposed to be plural. Sorry. Fixed now.

PR #197 (2020-11-21): Improve serial port detection
Fixes #192 .

Changes proposed in this request:
- Adds and gives user a command line option to use config.json to specify serial ports with default devices expected
- If not using config.json, uses pyudev to automatically detect adding and removing Arduinos (this is the default option)
- Does not enforce having 4 serial ports to open; uses the serial port count for arduino_count
- Adds some constants at the top of GUI.py and adds some docstrings
- Adds some writes to logFile but also adds some prints to the screen for serial ports
- Tries to be better about using  [... cut]
closed 2020-11-21 by streadgold