ISS-Mimic/Mimic #195: Kivy graphics set_texture list index out of range

dallasbrittany, 2020-11-05:
This error comes up frequently but does not crash the program. It's just annoying and might be slowing down the GUI and/or making buttons not responsive.

`Traceback (most recent call last):
   File "kivy/graphics/texture.pyx", line 789, in kivy.graphics.texture.Texture.bind
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/text/__init__.py", line 754, in _texture_fill
     self.render(real=True)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/text/__init__.py", line 695, in render
     return self._render_real()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/text/__init__.py", line 678, in _render_real
     self.render_lines(lines, options, self._render_text, y, size)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/text/__init__.py", line 656, in render_lines
     render_text(line, x, y)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/text/text_sdl2.py", line 54, in _render_text
     self._surface.render(self, text, x, y)
   File "kivy/core/text/_text_sdl2.pyx", line 129, in kivy.core.text._text_sdl2._SurfaceContainer.render
 IndexError: list index out of range
 Exception ignored in: 'kivy.graphics.instructions.RenderContext.set_texture'`

streadgold, 2020-11-07:
Tried to recreate this error but haven't had success. I remember getting it a ton though when we were filming

dallasbrittany, 2020-11-07:
> Tried to recreate this error but haven't had success. I remember getting it a ton though when we were filming

I get it when I have just opened the GUI and am trying to press buttons with the touch screen. I find that the buttons are frequently unresponsive. I tend to be impatient and keep trying repeatedly. I'm fairly certain this is correlated to the error, but I haven't tried to prove it yet.

ISS-Mimic, 2022-01-21:
haven't seen this in a while, closing for now

closed 2022-01-21 by ISS-Mimic