ISS-Mimic/Mimic #196: Improve setup instructions

dallasbrittany, 2020-11-05:
[This setup page](https://github.com/ISS-Mimic/Mimic/wiki/Raspberry-Pi---Mimic-Software-Setup-Instructions) seems to be out of date.

- [ ] "Expand Filesize" should probably be "Expand Filesystem"
- [ ] This broke things for me: `curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -`
- I had to go back and remove what was added to `/etc/apt/sources.list.d` to make npm install work because of version issues.
- It should probably be `curl https://www.npmjs.com/install.sh | sudo sh` instead because that's what worked for me
- Previously, it was installing npm 5.8.0, but what I ended up doing installed 6.14.8. I also ended up with nodejs version 10.21.0
- [ ] Do we really need the `sudo` part of `sudo npm install lightstreamer-client`? It's generally not good to do `sudo npm install` of packages
- [ ] It's not clear if it's because of the previous curl issues or if it was something else, but I had to remove matplotlib and reinstall it (with `python3 -m pip`)

It would be best to try this on a clean install with the new install.sh file and see if it works smoothly. There may be more to do to update the instructions.

dallasbrittany, 2020-11-13:
Okay, I think I have things working with a fresh install. I can actually do serial write commands now; whereas, I couldn't before.

#### Summary of important changes
- I'm going to recommend the Raspberry Pi Imager because it's super easy to use, and then you don't have to download a Raspbian image separately
- Expand Filesize is actually Expand Filesystem
- I might add some other useful tips like how to install an on-screen keyboard, etc.
- Instead of using apt-get, I'm just using apt because [apt is supposed to be simpler](https://itsfoss.com/apt-vs-apt-get-difference/).
- I got rid of all the sudo commands in front of pip3 installs and npm installs. I am also not doing chown -R for the node_modules folder. My guess is the sudo for that was required because we did sudo install before. Unless there's a really important reason I'm not aware of, we really shouldn't sudo install more than we have to.
- To install npm, they first need to sudo apt -y install nodejs sqlite3. (Okay, sqlite3 probably isn't required first, but nodejs is, and I am combining similar commands on the same line that can be run at once.) Then to get the right version of npm, they need to run `curl https://www.npmjs.com/install.sh | sudo sh` because using apt will not work (if you try, you get npm WARN npm npm does not support Node.js v10.21.0)

Do you see any problems with this? If someone wants to run through a fresh install and test with the real system, I can paste all my notes here so you can literally copy and paste what I did. I already have it all written out. Otherwise, I can just edit the Wiki, and we can go from there. It definitely doesn't install as is, so we won't lose anything if the instructions still need tweaking later.

streadgold, 2020-11-13:
Those all sound good to me, feel free to go ahead and edit the wiki.

Side note - if you've done a clean install following those instructions and don't have any github credentials it might be good to copy that so we can update our source forge image with that too

PR #200 (2020-11-14): [#196] Improve setup
Fixes #196.

Changes proposed in this request:
- Adds setup script
- Updates README with new URL of setup wiki

@ISS-Mimic
closed 2020-11-14 by streadgold