ISS-Mimic/Mimic #202: Arduino Motor shields -- solderless solution

juggler2000, 2020-12-02:
To minimize complexity, we prefer our build to be possible without soldering.  This has mostly been achieved, but our beloved Adafruit motor shield v2.3 does not come with headers in place (and note that we need stackable headers, which is not what comes with it).  We also need to set one of the solder jumpers so that the board uses 3.3v logic instead of the default 5v. If we don't do this, we could fry some of the pins on our Arduino (or maybe even the entire board). 

Since the board is open source (I think), one possibility is to make our own design of this board (in Gerber or similar), so that it could be ordered from a PCB manufacturer as an assembled, ready-to-go piece. We use three of these in our build. 

Adafruit Motor Shield: https://www.adafruit.com/product/1438

streadgold, 2020-12-02:
Alternatively..... Combine the motor shield and hdmi shield into one design... Stretch goal

juggler2000, 2020-12-02:
Yea, that's even better. And that way we could have our own connectors for motor power, encoder power, sensor data returning, etc.  That would be very, very nice.

cactus314, 2021-07-28:
The current solution for reducing the soldering required for these shields is to use clones of the Adafruit motor shield off of Amazon: [HiLetgo Motor Shield](https://smile.amazon.com/HiLetgo-TB6612FNG-Stepper-Motor-Replace/dp/B073SP76MC/ref=sr_1_2?dchild=1&keywords=hiletgo+motor+shield&qid=1627482601&sr=8-2).
They come with the stacking headers already soldered in place, plus they're cheaper (we're just not supporting Adafruit).  Preparing them for use on the Adafruit Metro 0 (3.3V logic) is a 5-minute process once you have your tools together and iron heated up.  I'm preparing the documentation and video for the conversion steps.

SpaceTowel, 2021-08-11:
BOM updated with Motor Shield.  Shield is being tested.

closed 2021-08-11 by SpaceTowel