ISS-Mimic/Mimic #213: Is the ISSLIVE telemetry offline?

connorjak, 2021-01-17:
I noticed yesterday and today the example implementation here: https://iss-mimic.github.io/Mimic/dashboard.html is currently `NO DATA` for all data channels. The official ISSLIVE site appears to also appears to be `NO DATA` as well. Has the telemetry been offline for a while, has this happened before, and do we have any contacts in Flight Ops who would know more details?

connorjak, 2021-01-17:
Duplicate of #100; should close this issue when/if that one reopens.

ISS-Mimic, 2021-01-17:
Hi connorjak,

Yep data is offline due to the Solarwinds cybersecurity incident. It was supposed to come back online this past week but obviously that did not happen. We do know the team that manages the ISSlive server and we'll ask them for a status again if it's still offline next week.

connorjak, 2021-01-17:
Is there an easily accessible repository of historical telemetry, or at least datasets with a few hours of telemetry? My research lab is planning to use my realtime ISS simulation for VR human-in-the-loop space mission research, so having a good backup option in place of the realtime data would be great. It wouldn't need to be another lightstreamer server; my current software setup is an abomination of NodeJS and hastily-coded C++ socket IO so I'm looking to replace it anyway.

ISS-Mimic, 2021-01-17:
Yep, well kind of. We record all of the ISS telemetry when available. I upload it for the ISS national lab folks but feel free to use it yourself. I think just sharing the link should allow you to download it.

https://www.dropbox.com/sh/tzc10hhlmhahcyk/AABPDWZitIwkkGgZv-hN0OxOa?dl=0

connorjak, 2021-01-17:
Great, this is exactly what I had in mind. I take it the timestamp on each data point is `Hours past J2000 epoch`/`24`?

ISS-Mimic, 2021-01-17:
Oh yeah timestamp is a little weird, it's hours past midnight Jan 1 UTC of the current year, resetting every year

connorjak, 2021-01-17:
Well good thing I have https://github.com/HowardHinnant/date to take care of the leapseconds; I should be able to handle that cleanly.

closed 2021-01-17 by connorjak