ISS-Mimic/Mimic #222: Motor issue

streadgold, 2021-07-17:
We keep burning out metros on the starboard side. Due to issues turning the 3a motor manually I am guessing maybe there is a problem with that motor. Unknown of it is causing the metro to burn out

cactus314, 2021-07-18:
If it's always on the starboard side, have you also always been using the same motor driver shield?  It could be bad somehow.  I looked today and verified (visually and by probing) that the Vin jumper isn't shorted and that the 3.3V logic is properly bridged and the 5V logic connection is severed/isolated.  Not sure what else to check except just to swap it out and see if the Metros stop burning out. 
[image]

juggler2000, 2021-08-01:
Maybe mark this driver board with a "sus" and replace it as @cactus314 suggested.

SpaceTowel, 2021-08-04:
Replace Motor shield?   Patrick to Test Motor shield on an isolated Arduino/board.
Slip ring?  ways to debug?   
Very hot at the voltage regulator at time of burn out, no LED (aka, no 5V).   hook up leads to burn board to see if it still regulate voltage. 
Sam to ask electrical specialist on ways to figure out how to troubleshoot.

cactus314, 2021-08-06:
@streadgold - Just to verify, this is the component that is getting hot on the Metro 0, correct?  I may throw a thermocouple on it while measuring various motor currents from the motor shield to see if there's a correlation to any specific BGA motor on the starboard side,
[image]

streadgold, 2021-08-06:
I believe so. Was in that area for sure

cactus314, 2021-08-06:
@juggler2000 @streadgold 
Spent some hours debugging the motor situation today.  tl;dr:  ¯\_(ツ)_/¯ 

- free-running geared motor out of the box pulls ~65mA at 12VDC; ~95mA when finger pressure is applied to the shaft (not stall)
- Prepared the Arduino IDE for the Metro 0 chipset, including loading the encoder.h library (hopefully the correct one).  Tested by loading a 'blink' sketch to a new Metro 0 and everything worked, so I pulled the Ard_StbdBGAs_r18 .ino files and uploaded them to the Metro 0.
- Taped a small thermocouple to the power regulator chip on the Metro to see if I could monitor/identify thermal effects.  **NOTE**: During the course of these tests, the maximum temperature increase was 1degC, so I didn't register any thermal failures.
- Used the 'old' motor shield and Mimic shield as rewired when the STEM Mimic was brought to my house.  One lead from each motor was disconnected from the Mimic shield to prevent any motors from turning when power was applied to the shield.
- Connected the Metro 0 to the proper USB cable, applied 12VDC to the motor shield, but did not begin to run GUI.py
- As soon as I put a current meter in series with Mimic Shield M1, BGA 3B began turning.  Current draw was ~60mA.
- Tested Mimic Shield M2, BGA 1B began turning.  Current draw was 65-73mA.
- Tested Mimic Shield M3 - no spontaneous motion of any BGA motors.  Ran GUI.py, but was unable to cause any BGA motion on the starboard side.  (Port side appeared to respond appropriately.)
- Closed GUI.py, tested Mimic Shield M4, BGA 1A spontaneously turns.  Current draw ~60mA.  Noticed two of the Neopixels on the 1A IEA board were on.
- Since the Mimic Shield M3 connections didn't cause motion (presumably of BGA 3A), I moved those connections to Mimic Shield M4, but this caused BGA 1A to spontaneously begin turning again.  Curious.
- So, plugged all motors back into original positions, GUI.py not running.  BGAs 1A & 3B rotate, 1B oscillates, 3A static.  By running GUI.py, it seems like I can cause small temporary motion changes to the BGAs, but then they resume previous motion.  I cannot get 3A to move.  BGA 1A reversed direction during this portion of the test.
- Disconnected the USB cable from the Metro 0 and powered the Metro via 9VDC on the barrel jack.  This was to prevent any spurious serial data that might have been coming in over the USB.  Similar spontaneous behavior as before, though.
- Installed a new motor shield (modified for 3.3V logic).  Similar spontaneous behavior.
- Uploaded a 'null-code' for firmware onto the Metro 0.  No response from motors.  Reinstalled Ard_StbBGA_r18 and behavior resumed.

At this point, I don't know if this is a software issue or maybe a bad Mimic shield?  I intend to swap out the Mimic shield with one of the new ones that I got in from JLCPCB, but I'd like to know what the status LEDs mean.  But this feels like software and I'm kinda afraid it's something simple like my not using the IDE properly.  In any case, I couldn't diagnose any reasons for killing Metros on the starboard side, I just observed weird behavior.

If you've read this far, congratulations.  Thoughts and advice appreciated.

cactus314, 2021-08-07:
Checked coil resistances of motors this morning:
Out of the box, geared motor resistance ~ 19 Ohms
Measured across Mimic Shield M1: 19 Ohms
Measured across Mimic Shield M2: 14 Ohms
Measured across Mimic Shield M3: Open circuit
Measured across Mimic Shield M4: 19 Ohms

Isolated BGA Motor 3A using a separate breakout board (one of the new ones).  Motor coil resistance: ~34 Ohms.  Applied 12VDC, and the motor turned, pulling 130-150 mA (compared to ~60 mA for the other motors).

So there appears to be a break in the BGA 3A circuit, and the motor might be failing.  My eyesight is getting worse, so even with glasses I was unable to test the existing HDMI breakout board to see if the problem was there, and I don't know the pinout of the existing IEA breakout board to test anything there - plus the whole eyesight thing.  I need new glasses and/or a hand-held magnifier.

cactus314, 2021-08-14:
Troubleshooting update:

- "Jiggling" the 1A/3A HDMI breakout board fixed the open circuit on BGA 3A and now reads 19Ohms through the motor coil
- The starboard Arduino Motor Shield Stack had been installed and labled backwards when delivered (i.e., 1B/3B HDMI running to 1A/3A BGA motors)
- Error in the Serial Output debug section of the Ard_StdbBGAs_r18 code for displaying the proper encoder output - fixed locally, but not uploaded to GH yet
- The exisiting Mimic Shield was replaced with one of the new Mimic Shields from my July order in the orientation specified by the silkscreen and the encoders responded appropriately to manual rotation of the individual motors
- The motors were wired according to the silkscreen on the Mimic Shield, but the erratic motor rotations/oscillations continuted (presummed mismatch between motor output and encoder feedback)
- Motors re-wired to match motors to appropriate encoders (Motor Shield M1-> Mimic Shield M2; Motor Shield M2->Mimic Shield M1; Motor Shield M3->Mimic Shield M4; Motor Shield M4->Mimic Shield M3)
- Now on power-up, all BGAs respond with appropriate initialization move and then stop, though there is some slight hum from at least one motor.
- Was going to test the motor control from the RPi GUI, but the display won't come on.  Opening a ticket for that and will troubleshoot it next (likely a loose connection in the display ribbon cable?)
- I have not yet changed out motor 3A as @streadgold requested.  I'll do that soon.

@streadgold - Not sure if there is a problem with the Mimic Shield or somewhere in the HDMI breakouts that is requiring me to swap the motor connections.  Also, there appears to be a problem (intermittent) with the 1A/3A HDMI breakout board causing open circuits.

**Finally, none of this necessarily explains why starboard Metro BGA controllers keep failing.**

juggler2000, 2021-08-15:
oof.

Had any similar wire tracing issues on the port side?

cactus314, 2021-08-15:
@juggler2000 - The port side has been working like a champ (BGAs & IEA Neopixels) ever since it has arrived here at my place.  Now I've probably jinxed it...

SWGlassPit, 2021-11-15:
Update: putting a scope on the assembly shows that the 3.3v rail is getting pulled down to -0.8v (!) at the precise moment the motor lines get energized. This happens even when the HDMI cables (including the motors) are disconnected. Sending 3v3 below ground would certainly fry the regulator or microcontroller. Appears to be capacitive coupling somewhere.

Additional troubleshooting steps to try:
 [ ] Verify whether this effect is isolated to the 3.3v rail or whether the 5v rail is affected as well
 [ ] Determine if the effect persists when the wires from the motor driver shield to the mimic shield are disconnected (i.e., is the coupling in the traces of the mimic shield or elsewhere?)
 [ ] If yes, does the effect persist when the mimic shield is removed entirely?
 [ ] Else, is the effect in any way related to the loose wires between the motor shield and the mimic shield? Does twisting them help?

Potential workarounds:
 - additional bypass capacitors on mimic shield between power rails and ground
 - adjust wire routing (and possibly HDMI pinout) to keep ground traces between motor traces and power rails
 - add reverse bias protection diodes (or some other solution) between power rails and ground to keep the reverse polarity clipped to reasonable levels. Schottky diodes will limit to -0.3v. (note this may be a red herring -- while reverse polarity protection isn't a bad idea, far better to eliminate the source if possible)

ISS-Mimic, 2021-11-16:
[image]

@SWGlassPit Yeah the 3V and 5V lines are in the adjacent pins to one of the motors. I wonder if disconnecting the motor that runs next to those pins and then scoping the other motor and 3V would clue us into if the HDMI pin selections is the issue?

SWGlassPit, 2021-11-16:
That's not a ton of overlap between traces. I'm wondering if the behavior
exists if we pull the mimic shield off but leave the motor shield on

SWGlassPit, 2021-11-17:
Another possibility I had not previously considered may be that the motor driver is pulling ground up instead of the rails down if our grounding strategy tying the two together is ill advised.

juggler2000, 2022-11-30:
Since it hasn't happened in a year, we must have fixed it! :P  Closing issue to tempt fate.

closed 2022-11-30 by juggler2000