ISS-Mimic/Mimic #223: Import and Runtime errors with numpy and missing 'service_identity' module

cactus314, 2021-07-17:
After manually installing the Mimic software (mostly following the Wiki guide), the following errors show on exit:pi@raspberrypi:~/Mimic/Pi $ python3 GUI.py
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_21-07-17_6.txt
[INFO   ] [Kivy        ] v2.0.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 186 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: sdl2(['text_pango'] ignored)
Mimic Directory: /home/pi
Serial ports opened: []
Started monitoring serial ports.
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'2.1 Mesa 19.3.2'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'V3D 4.2'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <b'1.20'>
[INFO   ] [GL          ] Texture max size <4096>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Clipboard   ] Provider: xclip
[INFO   ] [CutBuffer   ] cut buffer support enabled
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Loader      ] using a thread pool of 2 workers
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event3
[INFO   ] [MTD         ] Read event from </dev/input/event3>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [MTD         ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event4
[INFO   ] [MTD         ] Read event from </dev/input/event4>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [MTD         ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event3
[INFO   ] [HIDInput    ] Read event from </dev/input/event3>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [HIDInput    ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event4
[INFO   ] [HIDInput    ] Read event from </dev/input/event4>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [HIDInput    ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event3> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event4> range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event3> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event2> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event4> range position Y is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] using <mini keyboard Consumer Control>
[INFO   ] [MTD         ] </dev/input/event1> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] using <mini keyboard Mouse>
[INFO   ] [MTD         ] </dev/input/event2> range position Y is 0 - 0
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts>
[INFO   ] [MTD         ] </dev/input/event4> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 0
[INFO   ] [HIDMotionEvent] using <mini keyboard System Control>
[INFO   ] [HIDMotionEvent] using <mini keyboard>
[INFO   ] [MTD         ] </dev/input/event3> range pressure is 0 - 255
[... 285 lines cut ...]
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/ticker.py", line 179, in <module>
    from matplotlib import transforms as mtransforms
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/transforms.py", line 46, in <module>
    from matplotlib._path import (
ImportError: numpy.core.multiarray failed to import
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "/home/pi/Mimic/Pi/NightShade.py", line 1, in <module>
    import matplotlib as mpl
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/__init__.py", line 107, in <module>
    from . import _api, cbook, docstring, rcsetup
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/rcsetup.py", line 26, in <module>
    from matplotlib.colors import Colormap, is_color_like
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/colors.py", line 82, in <module>
    from matplotlib import _api, cbook, scale
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/scale.py", line 18, in <module>
    from matplotlib.ticker import (
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/ticker.py", line 179, in <module>
    from matplotlib import transforms as mtransforms
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/transforms.py", line 46, in <module>
    from matplotlib._path import (
ImportError: numpy.core.multiarray failed to import
Stopped monitoring serial ports.
[INFO   ] [Base        ] Leaving application in progress...
pi@raspberrypi:~/Mimic/Pi $

ISS-Mimic, 2021-07-20:
@cactus314 try "pip3 install --upgrade numpy" and see if the error returns

cactus314, 2021-07-22:
I couldn't recreate the error on my second (more successful) installation attempt.  When/if I try booting off the other MicroSD card again I will try this.  Otherwise, you can consider closing this issue.

cactus314, 2021-08-04:
I was using my "clean" install and got the numpy import error (must happen from some specific GUI window or sequence), so I tried the upgrade as recommended.  That killed everything and now GUI.py is fully broken.  See output below.  I'm going to redo the installation.
>>>>>>>>>>

pip3 install --upgrade numpy
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (1.16.2)
Collecting numpy
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.21.1-cp37-cp37m-linux_armv7l.whl (12.3 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 12.3 MB 4.8 kB/s 
Installing collected packages: numpy
Successfully installed numpy-1.21.1
WARNING: You are using pip version 21.1.3; however, version 21.2.2 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
pi@raspberrypi:~/Mimic/Pi $ python3 GUI.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "GUI.py", line 13, in <module>
    import numpy as np
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/__init__.py", line 150, in <module>
    from . import core
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python3"
  * The NumPy version is: "1.21.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

SpaceTowel, 2021-08-04:
check the version of NumPy that encountered the error.

cactus314, 2021-08-18:
@streadgold - Per recommendations at: https://numpy.org/devdocs/user/troubleshooting-importerror.html#raspberry-pi
I tried:
`sudo apt-get install libatlas-base-dev`
which fixed the numpy error and allowed GUI.py to run as normal without error.  This seems like a clean fix, but I'm not sure I should have upgraded numpy to begin with?  Consider closing this issue if you want.

streadgold, 2021-08-18:
Still disconcerting that upgrading numpy caused so much trouble but oh well. 

I've added the apt install option to the software install page

closed 2021-08-18 by streadgold

cactus314, 2021-08-18:
@streadgold : The alternative solution was to use the Raspbian version of numpy:
```
pip3 uninstall numpy  # remove previously installed version
apt install python3-numpy
```

streadgold, 2021-08-18:
Yep that's what I used, I like the apt install option best