ISS-Mimic/Mimic #226: Add Micro gear dc motor screws to BOM

Digisembler, 2021-07-21:
Went to go fix the S3 motor mount issue #204 and didn't have the nominal screw length, and it needs to be added to the BOM regardless

juggler2000, 2021-08-11:
M1.6x 0.35, 5mm length, hex head. Will look for a vendor.

juggler2000, 2021-08-12:
This seems like a good option. Countersunk, with torx head. https://www.mcmaster.com/92703A142

cactus314, 2021-08-12:
> This seems like a good option. Countersunk, with torx head. https://www.mcmaster.com/92703A142

Do the outboard truss pieces also need to be modified to accommodate that countersink?
Should I order some and try them on my build before adding them to the BOM?

juggler2000, 2021-08-12:
Utilized same screws as ordered for PIRB unit.  Updated BOM to add item and reflect updated costs.

closed 2021-08-12 by juggler2000