ISS-Mimic/Mimic #232: Errors in the detailed frame build guide

cactus314, 2021-07-24:
I finally built out the Mimic frame following the detailed build guide and found a few details which aren't consistent with the current STEM model and frame BOM.  Most notably, the guide calls for small L-brackets in many places where large brackets are used.  There are also some discrepancies (known and noted in the guide) in the dimensions between the vertical elements of the frame.  I took a number of photos during the build process and can take more of the finished frame (without all the cabling, etc.) as required.  The photos are currently on a GoogleDrive, but I can share them in whatever way is best.  I'm happy to help edit the build guide in whatever capacity in order to make it current and consistent with the STEM model and future builds.
[image]

Digisembler, 2021-07-24:
Thanks @cactus314 !  I was noticing on the 2nd build we're working on that we might end up with side mounted L brackets to help stiffen the short vertical pieces holding up the pressurized modules, as ours are pretty floppy. Cannot use T plates as they're too close to the main central brackets.  I think we also had to remove the back central bracket to get correct spacing between the node 1 and node 2 vertical mount rods. Let us know how it comes along for you please!

juggler2000, 2024-02-01:
This one is now redundant to this: https://github.com/ISS-Mimic/Mimic/issues/512 .

Also, ppt from 2022 seems to have all the of the updates: Wiki/Mimic One Frame Updated.pptx

closed 2024-02-01 by juggler2000