ISS-Mimic/Mimic #233: Error in Pre-Assembled PCB costing

cactus314, 2021-07-25:
The sub-total listed for the (mostly) pre-assembled PCBs under [Custom Printed Circuit Boards (PCBs) - Assembled](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Bill-of-Materials#custom-printed-circuit-boards-pcbs---do-it-yourself-diy) appears to be for the boards from JLCPCB only and doesn't include the extra parts needed from Amazon to be soldered to the underside of the boards - should be an extra $26.78 or so.  This might impact the [final totals](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Bill-of-Materials#mimic-total-cost) below as well - I didn't check.

SupaTreadz, 2021-07-25:
ah! good catch thanks

streadgold, 2021-07-25:
ah! good catch, thanks

streadgold, 2021-07-25:
Also added HDMI cable cost, now the pre-assembled are more expensive than DIY

closed 2021-07-25 by streadgold