ISS-Mimic/Mimic #237: Changes/Additions to the "Essential Parts" BOM

cactus314, 2021-07-29:
Suggested changes/additions to the [Essential Parts](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Bill-of-Materials#essential-parts) BOM:

1. The [Traxxas 3942 pinion gear](https://www.amazon.com/gp/product/B0006O5L1S/) is no longer available and has been superseded by the [Traxxas 3942X](https://www.amazon.com/Traxxas-3942X-Hardened-12-Tooth-Pinion/dp/B078KFCN72/ref=sr_1_1?dchild=1&keywords=Traxxas+3942x&qid=1626536712&s=toys-and-games&sr=1-1) (Amazon).
2. Change the Adafruit Motor Shield to the [HiLetgo Motor Shield](https://www.amazon.com/gp/product/B073SP76MC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) (Amazon) which has the stacking header pins already installed (and is cheaper).  This shield still requires modifications to convert to 3.3V logic.
3. Need to add [M3 threaded rod](https://www.grainger.com/product/FABORY-Fully-Threaded-Rod-25DM75) (1-meter, 2 each required) from Grainger to the BOM.  These are used for supporting the inboard truss sections.
4. Add [M3 nylocks](https://www.lowes.com/pd/Hillman-8-Count-3mm-Zinc-Plated-Metric-Nylon-Insert-Lock-Nuts/999996650) (4 required).  Probably cheapest to get them from a local Lowes/Home Depot/Ace for such a small qty.
5. Add [velcro cable straps](https://smile.amazon.com/gp/product/B00O9VKVFK/ref=ewc_pr_img_1?smid=A2UV2VXDDFSPSW&psc=1) (Amazon) - optional, but nice for cable management.
6. Consider replacing the [recommended jumper wires](https://www.amazon.com/EDGELEC-Breadboard-Optional-Assorted-Multicolored/dp/B07GD2RP9C/) (all pin-to-pin) with a package that also includes pin-to-socket jumpers that can be used with the Neopixels.  [This package](https://smile.amazon.com/gp/product/B01EV70C78/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1) is an option (Amazon), but the jumpers are only 20cm instead of 30cm as stated in the recommended BOM version.  I don't know how big a problem that would be when wiring the motor shields to the Mimic Shields.

juggler2000, 2021-08-01:
1. Unfortunately the new gear is almost 2x the price of the old one.  But I haven't looked for another option.  This is fine since we only use two.
2. Agree.
3-4: Might be nice to use either Grainger or McMaster for the specialty smooth and threaded rods, rather than buying from both.  Could also include the M3 nylocks as an option there, as they seem to be available for <$4 for 50, the smallest quantity I saw.   Priced similar at Amazon, but cannot get the desired rods there.
5. Yes!
6. The current custom PCB Shield requires routing across the board to reach the proper screw terminals on the Motor Shield.   It seems like we are using pretty much all of the length of the current jumpers, but perhaps you can measure to confirm.
7. Metro Mini is fine with me.  Whatever board we use, just need to ensure we don't accidently send 5V to any 3V boards.  I think we could still run the Mini at the default 5V logic, but just need to double check this stuff.  5V preferred for LED power so we can go brighter, of course.  But I think that the IEAs (by theBGAs), we borrow 3V for the LEDs so those will be a bit dimmer.

cactus314, 2021-08-02:
Item 1: Not to quibble, but looking at the [existing BOM](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Bill-of-Materials#essential-parts), I think the price difference is only $1.19 each.

Item 3: Can't get the smooth rod from Grainger in a 2-ft length, and if we want the threaded rod to be stainless, it's cheaper (when considering shipping) to source two suppliers and order the threaded rods from Grainger.  I suggest that we go with [unplated steel from McMaster](https://www.mcmaster.com/98861A040/) for the M3 threaded rod since it shouldn't be exposed to the elements, then the 2-ft 3/16" rods, M3 threaded rods and M3 nylocks can all come from McMaster and get us the cheapest price when shipping is considered.  We just sacrifice the stainless material choice for the all-thread.

Item 6: I found a multi-pack with 30cm jumpers.  I do agree with you that you need the full 30cm for the motor hookups.  So long as you don't need more than 40 pin-to-pin jumpers total, [this pack](https://www.amazon.com/EDGELEC-Optional-Breadboard-Assorted-Multicolored/dp/B07GD2PGY4/ref=sr_1_3?dchild=1&keywords=30cm%2Bfemale%2Bjumper%2Bwires&qid=1627903616&sr=8-3&th=1) from Amazon should work, and includes the required socket jumpers for the Neopixel connections on the Mimic shield as well.

Item 7:  I thought you were already using a 5V-logic Arduino for the Neopixels.  I don't know for sure because that setup didn't make it to my house during the rapture, so I can't inspect the way y'all had the Neopixels hooked up except to know that they're getting power from the 3.3V Metro boards along with the encoders.  Using 5V logic and 3.3V power doesn't seem to be a problem so far as I can tell from a cursory Google search.

juggler2000, 2021-08-02:
1) agree.
3) okay!  sounds good.
6) Sweet.  
7) You are right.... obviously it was 5V logic  (in hindsight) since we were using a Mega and a Pro Mini.

closed 2021-08-03 by cactus314