ISS-Mimic/Mimic #244: Need to create a current Raspberry Pi image file for easy download and install using RPi Imager

cactus314, 2021-08-01:
Need to update the .img file linked in the [Mimic Software Setup Instructions](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Mimic-Software-Setup-Instructions#option-1---pre-configured-raspberry-pi-mimic-image) on the Wiki.
This might be useful:
[How to Back Up Your Raspberry Pi as a Disk Image](https://www.tomshardware.com/how-to/back-up-raspberry-pi-as-disk-image)

SpaceTowel, 2021-08-04:
Patrick will pass the clean .img file to Sam (be sure to remove wifi pass code, and Github credentials), Sam will update source code.

cactus314, 2021-08-09:
Should I include the keyboard app and figure out a desktop icon (per Issue #171) before making the image?

ISS-Mimic, 2021-08-09:
Definitely include the keyboard app, and if you can make the desktop icon work sure. I remember having relative directory issues when working on the desktop icon but that might not be a problem now that the code doesn't have hard coded paths

cactus314, 2021-08-11:
> Definitely include the keyboard app, and if you can make the desktop icon work sure. I remember having relative directory issues when working on the desktop icon but that might not be a problem now that the code doesn't have hard coded paths

I have the keyboard app installed with a taskbar toggle and have the single-click desktop icon working as well - though I haven't strenuously stress-tested it. I did leave it in terminal mode just so folks would see something happening for the few seconds it takes for the main Mimic GUI to open on the screen.  I think this should close Issue #171.
[image]

Haven't yet made a clean disk image for @streadgold to upload.  I'll get to that soon.

cactus314, 2021-08-22:
Was successful in creating a .img file useable by the RPi imager program to create a bootable MicroSD card, but there are multiple "embedded null byte" errors when trying to start GUI.py.  All of the basic RPi desktop functionality seems to work fine, and the Mimic desktop icon and keyboard taskbar toggle I installed are present.  Will try to create a fresh image and then troubleshoot from there.

cactus314, 2021-08-22:
I have a successful .img.gz image file that can be transferred to a MicroSD card using the RPi imager program.  The file is ~ 4.3GB (?).  I'll find a way to sneakernet it to @streadgold on a thumb drive, then this issue can be closed until major changes are made to the RPi program and we need to do it again.

closed 2022-10-20 by ISS-Mimic