ISS-Mimic/Mimic #245: Need to add shaft coupler set screws used on SAWs to BOM

cactus314, 2021-08-02:
What size thread and what length is used to replace the small setscrews that come with the couplers?

ISS-Mimic, 2021-08-04:
Is this just for if the screws get lost? The set screws should come with the couplers right?

cactus314, 2021-08-04:
I thought @juggler2000 told me that y'all replaced the originals with longer set screws for... reasons.  Maybe I misunderstood.

juggler2000, 2021-08-04:
yes, we use longer screws. The thought was to help retain the coupler within the solar array wing. Might be overkill. This is the rediculous screw set we bought, but I don't recall the size screw we used.

Hilitchi 260pcs M3/4/5/6/8 Allen Head Socket Hex Grub Screw Set Assortment Kit with Plastic Box 12.9 Class Black Alloy Steel https://smile.amazon.com/dp/B014OMFU1I/ref=cm_sw_r_apan_glt_fabc_7WH2CD7T6AQXY4PDRB6X?_encoding=UTF8&psc=1

SpaceTowel, 2021-08-04:
Patrick to remove set screw from Mimic and do measurement, and update BOM.

cactus314, 2021-08-07:
The set screws that come with the couplers are M3 x 3mm.  The set screw I removed from the model was an M3 x 6mm.
Recommend BOM item from McMaster: [Stainless Steel Flat-Tip Set Screw](https://www.mcmaster.com/92605A100/) rather than an assortment set from Amazon.  I'll update the BOM with these.

closed 2021-08-07 by cactus314