ISS-Mimic/Mimic #246: Spec out 2-ft HDMI cables instead of 3-ft cables

cactus314, 2021-08-04:
The cables listed in the [HDMI Cable BOM](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-Bill-of-Materials#hdmi-cables) are for a 3-pack of 3-ft cables (qty 2).  I didn't pay attention and ordered them, but they're really too long.
Suggest changing to [Cable Builders Short HDMI Cable 2-ft [2-Pack]](https://www.amazon.com/Cable-Builders-2-Pack-Ethernet-18GBps/dp/B07LFN7CX6), qty 3.  It's a couple of dollars more overall but removes having HDMI cable looping out from the frame.

I plan to return the ones I ordered and replace them with 2-ft cables.

ISS-Mimic, 2021-08-04:
Yeah good idea, dang I thought 3 ft was the good size but I remember going through this exact same thing

ISS-Mimic, 2021-08-04:
also sorry, I've been on vacation and then on work travel now so haven't been able to work on mimic stuff in a bit

closed 2021-08-05 by cactus314