ISS-Mimic/Mimic #250: New part order with new silk screen update from ticket 241 + 242

SpaceTowel, 2021-08-04:


cactus314, 2021-08-05:
@juggler2000 - Just to be clear.  You want me to use HDMI breakout boards and IEA Boards with the updated silkscreens on the build I'm currently working on?  That'll slow me down a little, but that's okay.  We have time.
@streadgold - What's the timeframe for updating the gerber files for those two boards?  Let me know when they're ready to go and I'll get the new boards on order.  I might have to go through the process of verifying the orientation of the screw terminals with JLCPCB again.

streadgold, 2021-08-05:
Should be able to do that tomorrow if I remember

cactus314, 2021-08-05:
> Should be able to do that tomorrow if I remember

Do you need dimensions for the offsets/terminal blocks?

streadgold, 2021-08-05:
Nope I already have the dimensions

juggler2000, 2021-08-05:
Yes, I was thinking of using the revised boards in your design, for photos and later debugging purposes. If that's a significant slowdown for you, then just proceed with what you have. Sounds like Sam is on it, though.

cactus314, 2021-08-09:
@streadgold & @juggler2000 - On the HDMI breakouts, there's an inventory shortage of the header pins @ JLCPCB (presumably the ones for the servos).  I'm tempted to order anyway and solder in those pins myself just to keep things moving.  Thoughts?
[image]

cactus314, 2021-08-09:
Or is that not how it works and they'll hold production until the parts become available?

cactus314, 2021-08-09:
Now that I'm exploring the tool, I see you can change parts.  Is this an acceptable replacement?
[image]
[image]

cactus314, 2021-08-09:
Or basically any straight three-pin header?
[image]

streadgold, 2021-08-09:
Yes any 2.54mm pitch 3 pin header should work, that looks fine to me. 

To answer your first question I'm not sure if they would hold product. I think if you just uncheck the part it would probably go through and just ignore that part

cactus314, 2021-08-09:
Had a minimum order of 50pcs for the 3-pin headers... not sure why, but fortunately they're super cheap.

closed 2021-08-09 by cactus314