ISS-Mimic/Mimic #254: Map out magnet orientations for modules

cactus314, 2021-08-10:
Should there be a driving schema (e.g., "active" v. "passive" port), or just copy what's been done on the STEM Mimic model?  How is it being done on the Boeing model?

cactus314, 2021-08-11:
Began an Excel spreadsheet here: [ISS Mimic Magnet Mapping](https://github.com/ISS-Mimic/Mimic/blob/master/docs/ISS%20Mimic%20Magnet%20Mapping.xlsx).  So far it has components listed on the US segment, and no magnet polarities (yet).

SpaceTowel, 2021-08-11:
For Reference: https://historicspacecraft.com/ISS.html
For Reference: https://historicspacecraft.com/Diagrams/S/ISS_Visiting_Vehicles_10Apr2016_RK2016.jpg
Create views using Blender model.

cactus314, 2021-08-18:
I've updated my [magnet polarity spreadsheet](https://github.com/ISS-Mimic/Mimic/blob/master/docs/ISS%20Mimic%20Magnet%20Mapping.xlsx), and put polarities on a couple of diagrams, though these images don't capture all of the magnets (or irregularities - particularly on the US segment).
[image]
[image]

cactus314, 2021-08-21:
Updated and uploaded a new spreadsheet based on input from @Digisembler.  Do we need graphics as well, or does the spreadsheet (and later a similar table in the build guide) suffice?

cactus314, 2021-09-01:
The views from @Digisembler look good and I think they'll work fine.  Could I ask for two things:
- [ ] Remove the Russian VVs from the images - I think they add distraction, particularly from the top view looking down on the SM
- [ ] Include a port-side view in addition to the starboard-side view so I can capture Node3 and JEM better as well.  So, three images in all (top, port & starboard)

cactus314, 2021-09-08:
Thanks to @Digisembler for the new drawings.  Two more modifications (I think) to help make the magnet layout more readable:
Remove the airlock and COF from the starboard side view, and remove MRM-2, Z1 and PMA3/IDA3 from the top view.  That clears the way for placing the "N"/"S" markers more directly near the correct ports/CBMs without having to overlay them on top of the modules.  Thanks.
[image]

ISS-Mimic, 2023-12-09:
Closed - done in @GodspeedDiscvry 's files.

closed 2023-12-09 by ISS-Mimic