ISS-Mimic/Mimic #256: Continue to update Acronyms page

cactus314, 2021-08-11:
Update [acronyms page](https://github.com/ISS-Mimic/Mimic/wiki/Acronyms) to include acronyms used for ISS pressurized modules/components (i.e., PMA, COF, FGB, etc...).  I don't mind working on this - it will help drill these into my head.

cactus314, 2021-08-21:
I've added (sometimes with Wikipedia links) all the acronyms I've run across so far in the physical build and in the RPi GUI with the following exceptions.  I could guess at some of them, but I'll let someone else fill them in, then this issue can be closed.
- [ ] EPS
- [ ] GNC
- [ ] RSOS
- [ ] TCS

Digisembler, 2021-09-01:
Defined EPS/GNC and TCS, removed RSOS (left ROS already there) and Created ticket #268 to fix the GUI to remove RSOS.  Leaving this issue open to account for any more as we go.

closed 2022-01-21 by ISS-Mimic