ISS-Mimic/Mimic #257: Put discrete module STLs into their own 3d Printing directories

cactus314, 2021-08-11:
Right now there are some modules that share components (and thus directories).  It would be nice for them to each have their own directory so it's clear that the parts go together.
For example, Columbus and Leonardo share the same barrel but have different end caps - all stored in the Columbus directory.  If I just went searching for the Leonardo model, I wouldn't know where to look.
The same is true for Node 3.  In the spreadsheet on the[ 3D Printing page](https://github.com/ISS-Mimic/Mimic/wiki/Build-Instruction%3A-3D-Printed-Parts) of the wiki, it's recommended to just print a second copy of Node 2 to use for Node 3 (which would mean Node 3 would have the support rod hole), but instead, I suggest that the Node 3 directory have the optional STL models (barrel without the support rod hole and endcap) which are also stored in the Node 2 directory,

I am not suggesting that PMAs, IDAs and identical ITS truss elements necessarily need their own directories.  Hope this makes sense.

Digisembler, 2021-08-11:
So... Leonardo/PMM and Columbus are the same "can" (there were 4 cans built, 3 as MPLMs, 1 as Columbus).  Since the Columbus is the only one with the "back porches", the end cap is different.  I think on the build sheet, the can portion is to be printed twice, and then each end cap once for the respective module.
Node 2 and Node 3 are identical, and should be on the build page to print two each (if i created a new end cap without a support hole, we could put it there.  also Node 3 zenith cbm is deactivated and has a grapple in the center of the port, so, i could make a complete 2nd version).

SpaceTowel, 2021-08-11:
new issue to be created on cleaning 3D print table

cactus314, 2021-08-12:
> ... also Node 3 zenith cbm is deactivated and has a grapple in the center of the port, so, i could make a complete 2nd version).

For now, I'll just print a small disk to plug the zenith CBM on the model I'm building rather than put in a magnet or leave it open.

ISS-Mimic, 2023-08-30:
done for HiFi modules at least

closed 2023-08-30 by ISS-Mimic