ISS-Mimic/Mimic #259: Switching neopixle control board to metro mini

SpaceTowel, 2021-08-11:


cactus314, 2021-08-12:
This is really a repeat of Issue #187 : "Replace LED-driving Ard Mega with Nano or similar"

cactus314, 2021-08-14:
_Sooooo_,
I've been playing with the Neopixels on the STEM Mimic.  (They don't all work, but that's another ticket.)
The Adafruit Metro Mini (and apparently the regular Metro 328) uses a SiLabs CP2104 hardware USB to Serial converter for USB connectivity (see [technical details](https://www.adafruit.com/product/2590#technical-details)).  They claim that this is well supported, but I can't get the Mini or a Metro 328 to respond to serial commands from the RPi (i.e., the onboard Rx LED doesn't flash and the Neopixels don't respond to commands from the RPi).  If I try an old Sparkfun RedBoard or use an Adafruit Metro M0 Express (the ones used for the motors), they receive the serial data just fine.  Interestingly, the Metro Mini and Metro 328 do show up as added Arduinos on the controller count on the Mimic GUI, they just don't receive data.

**Options**: I can try an Arduino Nano clone (needs to come with the pins not soldered so I can put them in upside down), but they usually require a USB-Mini rather than the USB-Micro that the Metros use - it would be nice for all the cables to be the same,  Alternatively we could go with a full-sized Arduino Uno clone that has a friendly USB converter.  Perhaps the best thing is to just add a 4th Metro M0 Express.  It makes the programming environment more consistent for the user, simplifies the BOM, and doesn't add that much cost - plus we know they work,
What say ye, @streadgold & @juggler2000?

juggler2000, 2021-08-14:
Interesting. I'm fine with a fourth Metro M0 Express. Thanks for all the work!

streadgold, 2021-08-14:
I'd be curious about getting a nano to work - just from a form factor perspective. If it doesn't work right off the bat though wouldn't put too much effort into it

streadgold, 2021-08-14:
Or like a trinket or something micro USB based, agreed that mini USB would be an obnoxious requirement

cactus314, 2021-08-15:
@streadgold - If I were smarter, I might be able to get the RPi to talk to the CP2104 ([reference here](https://www.raspberrypi.org/forums/viewtopic.php?t=255114)) on the Mini, but that'll take me some time to work through.  I'll order a Nano clone just to see if it works.  I didn't think the Trinkets had the USB/serial com capability required.
Most folks probably have extra USB-Micro cables scattered about, but it would be a shame to have to add them to the BOM for completeness.

streadgold, 2021-08-15:
@cactus314 good point, the standard trinket does not have a serial port but the trinket M0 does and I believe the pro micro does as well

cactus314, 2021-08-17:
@streadgold - I think I can make the Trinket M0 work wrt taking commands from the RPi, but it only has a single ground pin.  I measured the current in the ground wire between the Mimic Shield and the Neopixel controller (the Metro M0 I'm using right now), and see a draw of at least 120mA, so I don't think I can cheat and use one of the digital pins on the Trinket M0 as a GND to sink that much current.  We'd have to have the users make their own jumper to run two GNDs (port and starboard) from the single pin.  Not sure we want to ask folks to make their own jumpers.
I got in some Nano clones too, but I haven't tried them yet.  I suspect they'll work, but I was hoping for the Trinket M0.
I'm not sure what you're referring to with the "Pro Micro".  There's a "[Pro Trinket](https://www.adafruit.com/product/2010)", but it doesn't have a Serial-USB converter.

cactus314, 2021-08-17:
I suppose I could set Pin 4 low in the code and solder bridge it to GND on the backside as a cheat.  I don't know what bad things might happen that way.  It _should_ work okay.
[image]

streadgold, 2021-08-17:
Ah yeah didnt realize it only had the one ground pin. Might be more trouble than it's worth.

This is a pro micro https://www.sparkfun.com/products/12640 (or the sparkfun version, there are lots of cheap clones out there).

cactus314, 2021-08-17:
Metro Mini won't work due to an insufficient number of ground pins.  Closing out this Issue.  Issue #187 remains open to work on changing to a smaller device.  (Will try a Pro Micro next.)

closed 2021-08-17 by cactus314