ISS-Mimic/Mimic #262: Need to upload STLs for Visiting Vehicles

cactus314, 2021-08-12:
The following VV STLs need to be uploaded to the appropriate Visiting_Vehicles 3D Printing directories:

- [ ] Dragon
- [ ] Starliner
- [ ] Cygnus
- [ ] HTV

cactus314, 2021-08-13:
@Digisembler - Something's wrong with the scaling on Cygnus.  It comes in around 1m across in the x-dimension.  If I scale it down to 1%, it looks about right, but I don't know what the actual dimensions should be.

Digisembler, 2021-08-14:
1% is correct, as the model is apparently full size.  Checking my 3mf file I used to print, i noticed i had already scaled it there, and forgot that when I exported it (which i redid for the upload).

Digisembler, 2022-10-23:
Got most of Starliner uploaded (SM is too big, even zipped). Added the attributes for Cygnus and Starliner and HTV.

closed 2023-08-30 by ISS-Mimic