ISS-Mimic/Mimic #264: STEM Mimic touchscreen not turning on

cactus314, 2021-08-14:
The RPi is on (I can see the light), but the display is not coming on.  I suspect a loose connection in the display ribbon cable, but nothing was changed/moved to cause a failure.  Is this a recurring problem?

cactus314, 2021-08-14:
The screen would not come on with the USB hub attached.  Unplugged the HUB, rebooted, the screen came on, plugged in HUB and all is good.  Not sure what's up, but closing this ticket unless it happens again.

closed 2021-08-14 by cactus314

streadgold, 2021-08-14:
Was the USB hub power on? I've seen the Pi lock up from too much USB current draw before. Haven't quite experienced what you mentioned before though

cactus314, 2021-08-15:
@streadgold - I don't recall.  But I do know that I occasionally get "low power" notifications on the STEM Pi, which seems strange since the hub is powered.