ISS-Mimic/Mimic #278: Create a "MidFi" version of the Z1 Truss?

cactus314, 2021-10-30:
@streadgold's new Z1 Truss model is great, but it requires 8 individual parts to be printed and glued together.  Maybe there's a middle ground where the LoFi version can have the CMG elements modeled along with the antenna booms and dishes that can be printed as a single print.
[image]

ISS-Mimic, 2021-10-31:
[image]

@cactus314 thoughts?

ISS-Mimic, 2021-10-31:
also you can just copy an image file into a comment and it auto uploads to github thats nifty

cactus314, 2021-10-31:
Looks great!  Did you emboss the MBM on the forward side?

ISS-Mimic, 2021-10-31:
Yep MBM is still there!

closed 2021-11-03 by ISS-Mimic