ISS-Mimic/Mimic #279: Upload STLs in Binary Format

cactus314, 2021-11-01:
It's been noted that some of the posted STLs in the 3D Printing directories download directly (preferred) and some display the ASCII file which has to be copied, pasted into an editor, and manually saved with a .STL extension.  I uploaded the same model generated in SolidWorks both as a binary STL and an ASCII STL, and the binary version downloads in the preferred method.
I propose that all future STLs be submitted in binary format and that the existing files get changed to binary when/if they are converted to MM units for ease of use.  Just my 2 cents.

ISS-Mimic, 2021-11-01:
Agreed, let it be so!

ISS-Mimic, 2023-08-30:
pretty sure this is done

closed 2023-08-30 by ISS-Mimic