ISS-Mimic/Mimic #280: Need to upload current HDMI breakout board holder STL

streadgold, 2021-11-14:
The STL for this part is not current. The current design has a cut out in the bottom for the HDMI connector

[image]

streadgold, 2021-11-14:
This is the part:

https://github.com/ISS-Mimic/Mimic/blob/main/3D_Printing/Mimic_Structural_Compoments/HDMI%20Breakout%20Mount.stl

cactus314, 2021-11-15:
> This is the part:
> 
> https://github.com/ISS-Mimic/Mimic/blob/main/3D_Printing/Mimic_Structural_Compoments/HDMI%20Breakout%20Mount.stl

That part works with the latest design of the HDMI breakout board and is what I am using on my current build.  I'm not sure it works with the earlier boards.  Also, it could use some improvement, but it works (mostly).

Digisembler, 2022-01-12:
Corrected one uploaded, old one deleted.

closed 2022-01-12 by Digisembler