ISS-Mimic/Mimic #334: Beginner Projects

abrahammiranda, 2022-07-17:
Research different projects (such as Arduino starter kits and such) for people who want to start learning about the stuffs leading up to Mimic.

abrahammiranda, 2022-07-19:
Inspired from a father asking at Comicpalooza about how to introduce his child into stuff like this. I am thinking about linking to Arduino Starter kits, although there are other options such as [kiwico](https://www.kiwico.com/mlp/smarter?utm_source=Smarter_Everyday&utm_medium=INFL&utm_term=YouTube) that offer something similar. Should I make a page on the Wiki? The other option would be to include it at the end of the README.md. Lmk

P.S. Also heard LEGO Mindstorms and another one, but I can't remember

ISS-Mimic, 2022-07-19:
Ooohh kiwico is a good find yeah that would have been good

closed 2022-07-23 by abrahammiranda