ISS-Mimic/Mimic #350: Increase diameter of motor mount screw holes.

juggler2000, 2022-12-12:
[image]

Digisembler, 2022-12-13:
I've found that if one uses a drill bit, this is immensely helpful to enlarge a hole that might be slightly undersized from printing on a potentially less than superbly tuned 3d printer/slicer combo. :)  I suggest removing the shown screw, measure the diameter of the screw, and use that size drill bit on both holes (possibly on all holes for these motors), then reinsert the motor and tighten down with the screws.  While not in the BOM, a set of wire gauge drill bits is super helpful, but likely a standard or jewelers set from harbor freight or amazon, etc. would be close enough to widen the holes.  Go slowly, as if you spin a drill fast into 3d printed parts, you create a lot of heat that could deform the part more than you want.  I also do believe that these holes are countersunk, so once widening the holes, using a countersink bit to clean up the countersink may also be helpful.

juggler2000, 2022-12-13:
Let's keep the issues open until solutions are in the build. guide and reflected in the BOM.

juggler2000, 2024-01-24:
@SpaceTowel , did you have this issue with the recent build?  @Digisembler thinks it may have been a printer settings issue (if that's the case, need to add words to check size/drill out or define print parameters).