ISS-Mimic/Mimic #378: Orbit screen divide by zero error

ISS-Mimic, 2023-03-11:
[image]

SWGlassPit, 2023-03-11:
Problem is here: https://github.com/ISS-Mimic/Mimic/blob/bb4f9ba3a4ceb9d0f2df3ce7a715bef9bc7db6ab/Pi/GUI.py#L1313

There's a race condition between nightshade and orbitmap, as they are separate processes spawned by timers. Divide by zero can happen when orbitmap tries to read map.jpg while nightshade is trying to write it.

SWGlassPit, 2023-03-11:
Unfortunately, the reload method has a void return type so there isn't necessarily an easy way to see if the reload was successful.

Couple options here:

1. restructure so that the map update happens in memory rather than going through file i/o. I haven't looked to see what this would take.
2. Use the image width being nonzero as an imperfect proxy to check that the reload was successful, add a spin loop to the updateorbitmap method to keep trying the reload until it is successful. Should probably have a limited number of attempts so that we don't get stuck in an infinite loop if the write fails.
3. Update nightshade to write the new image to a different file, use a symlink to point to the new location, and have updateorbitmap point to the symlink. This doesn't eliminate the race condition completely, but it cuts the window down.
4. Set up either a mutex or a file lock to prevent reload from running while nightshade is running. Must be handled with care to ensure we don't get dangling locks that will break the code. If we have access to process IDs put out by popen, this might work as the lock as well.
5. Gracefully handle the failure to reload by checking for the zero conditions before dividing.
6. Tweak the timing of the timers to reduce the frequency of races by making sure the time intervals used to call the two methods are relatively prime.

None of these methods are foolproof, but they should reduce the likelihood of occurrence. There may be a foolproof way I haven't thought of, and implementing several of these is probably a good idea.

ISS-Mimic, 2024-08-11:
fixed in VV branch, just lazy checking if the denominator is zero and ignoring if so. should work merging soon

closed 2024-08-25 by ISS-Mimic